AlbrechtL/welle.io #441: Update welle.io

Rootinator, 2019-10-20:
Hi,

How to update an existing welle.io installation to the latest version?

andimik, 2019-10-21:
Well that is a general question...

You could rename the folder (in case the new version doesn't work) and clone it again and build it again.

AlbrechtL, 2019-10-22:
@andimik Thanks for answering.

closed 2019-10-22 by AlbrechtL

Rootinator, 2019-10-26:
I tried to compile welle.io (on the Raspberry), but unfortunately this fails.
I use cmake and get the following message:

`pi@raspberrypi:~/wio/welle.io/build $ sudo cmake .. -DRTLSDR=1
-- Build type not specified: defaulting to release.
CMake Error at /usr/lib/arm-linux-gnueabihf/cmake/Qt5/Qt5Config.cmake:26 (find_package):
  Could not find a package configuration file provided by "Qt5QuickControls2"
  with any of the following names:

    Qt5QuickControls2Config.cmake
    qt5quickcontrols2-config.cmake

  Add the installation prefix of "Qt5QuickControls2" to CMAKE_PREFIX_PATH or
  set "Qt5QuickControls2_DIR" to a directory containing one of the above
  files.  If "Qt5QuickControls2" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:50 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/wio/welle.io/build/CMakeFiles/CMakeOutput.log".`

If I open qtcreator on Windows instead and compile it for the Raspberry, I will only get errors.

After pressing "Create / Create All" the following errors appear:

`D:\Git\welle.io\..\welle.io\src\backend\dabplus_decoder.cpp:19: In file included from ..\..\..\welle.io\src\backend\dabplus_decoder.cpp:19:0:
D:\Git\welle.io\src\backend\dabplus_decoder.h:34: Fehler: neaacdec.h: No such file or directory
#include <neaacdec.h>
          ^~~~~~~~~~~~
D:\Git\welle.io\..\welle.io\src\backend\dab_decoder.cpp:19: In file included from ..\..\..\welle.io\src\backend\dab_decoder.cpp:19:0:
D:\Git\welle.io\src\backend\dab_decoder.h:31: Fehler: mpg123.h: No such file or directory
#include "mpg123.h"
          ^~~~~~~~~~
D:\Git\welle.io\..\welle.io\src\backend\dabplus_decoder.cpp:19: In file included from ..\..\..\welle.io\src\backend\dabplus_decoder.cpp:19:0:
D:\Git\welle.io\src\backend\dabplus_decoder.h:34: Fehler: neaacdec.h: No such file or directory
#include <neaacdec.h>
          ^~~~~~~~~~~~
D:\Git\welle.io\..\welle.io\src\backend\dab_decoder.cpp:19: In file included from ..\..\..\welle.io\src\backend\dab_decoder.cpp:19:0:
D:\Git\welle.io\src\backend\dab_decoder.h:31: Fehler: mpg123.h: No such file or directory
#include "mpg123.h"
          ^~~~~~~~~~
D:\Git\welle.io\src\backend\decoder_adapter.h:38: In file included from D:/Git/welle.io/src/backend/decoder_adapter.h:38:0,
D:\Git\welle.io\..\welle.io\src\backend\dab-audio.cpp:28: from ..\..\..\welle.io\src\backend\dab-audio.cpp:28:
D:\Git\welle.io\src\backend\dab_decoder.h:31: Fehler: mpg123.h: No such file or directory
#include "mpg123.h"
          ^~~~~~~~~~
D:\Git\welle.io\src\backend\decoder_adapter.h:38: In file included from D:/Git/welle.io/src/backend/decoder_adapter.h:38:0,
D:\Git\welle.io\..\welle.io\src\backend\dab-audio.cpp:28: from ..\..\..\welle.io\src\backend\dab-audio.cpp:28:
D:\Git\welle.io\src\backend\dab_decoder.h:31: Fehler: mpg123.h: No such file or directory
#include "mpg123.h"
          ^~~~~~~~~~
D:\Git\welle.io\src\backend\decoder_adapter.h:38: In file included from D:/Git/welle.io/src/backend/decoder_adapter.h:38:0,
D:\Git\welle.io\..\welle.io\src\backend\dab-audio.cpp:28: from ..\..\..\welle.io\src\backend\dab-audio.cpp:28:
D:\Git\welle.io\src\backend\dab_decoder.h:31: Fehler: mpg123.h: No such file or directory
#include "mpg123.h"
          ^~~~~~~~~~
D:\Git\welle.io\..\welle.io\src\backend\dabplus_decoder.cpp:19: In file included from ..\..\..\welle.io\src\backend\dabplus_decoder.cpp:19:0:
D:\Git\welle.io\src\backend\dabplus_decoder.h:34: Fehler: neaacdec.h: No such file or directory
#include <neaacdec.h>
          ^~~~~~~~~~~~
D:\Git\welle.io\src\backend\dab_decoder.h:31: Fehler: mpg123.h: No such file or directory
#include "mpg123.h"
          ^~~~~~~~~~
D:\Git\welle.io\src\backend\dab_decoder.h:31: Fehler: 'mpg123.h' file not found
D:\Git\welle.io\src\backend\dab_decoder.h:33: Fehler: "At least version 1.14.0 (API version 36) of mpg123 is required!"
D:\Git\welle.io\src\backend\dab_decoder.h:44: Fehler: unknown type name 'mpg123_handle'
D:\Git\welle.io\src\backend\dab_decoder.h:32: Warnung: 'MPG123_API_VERSION' is not defined, evaluates to 0`

andimik, 2019-10-26:
Indeed, it's quite not easy

I've only built welle-cli.

tenzap, 2019-10-27:
Have you installed all necessary build dependencies ?