AlbrechtL/welle.io #442: [macOS] Create DMG image

AlbrechtL, 2019-10-22:
welle.io runs also on macOS but it is not easy to compile and install it specifically for non experts.

It would be really nice to have a DMG to run welle.io.
Qt has also a deploy tool for macOS (https://doc.qt.io/archives/qt-5.11/osx-deployment.html).

The Linux AppImage is build with Travis CI. Luckily Travis support also macOS (https://docs.travis-ci.com/user/reference/osx/). So it is possible to create the DMG automatically like the Windows installer and the Linux AppImage.

@tenzap Are you interested in creating a DMG image?

tenzap, 2019-12-23:
If I understand well, according to this discussion [comment17](https://trac.macports.org/ticket/59779#comment:17), it seems welle.io cannot be distributed as binary on macOS because of licensing issues : one file of mpg123 (`./src/libout123/modules/coreaudio.c`), probably used for audio output on macos, is licensed under strict GPLv2.

AlbrechtL, 2019-12-27:
The `libout123` is only used for welle-cli but not for the GUI. If we distribute only the GUI variant there should be not problem.

PR #556 (2020-05-29): macOS DMG image build
This pull request adds a macOS build to Travis CI, which creates a DMG image for easy deployment and installation. This resolves #442.

Due to the licensing concerns raised in the issue this builds both the GUI and CLI, but packages and deploys only the GUI.

This also corrects a couple other minor issues with the Travis builds:

- Removes the deprecated sudo tag
- Fixes the CLI CMake build to build only the CLI, previously this was running a duplicate build of the GUI as well as the CLI

Bintray deployment has not been tested but I think it should work. Let me know if it [... cut]
closed 2020-07-12 by AlbrechtL