AlbrechtL/welle.io #452: [Linux Mint 19.2 / AppImage] Letters not readable

ontheair81, 2019-10-26:
Hello,
since an update of Linux Mint the letters in welle.io are not readable anymore. See picture:

[image]

This happens in welle.io 2.0, as well as in beta2 and beta3. Using the AppImage.

The problem just occurs after Linux Mint was updated yesterday. Here are all the softwares that have been updated: [updates.txt](https://github.com/AlbrechtL/welle.io/files/3774495/updates.txt)

Maybe this issue is in some way related to [#418 ](https://github.com/AlbrechtL/welle.io/issues/418) as it also happens in Linux Mint, after an update, also using the AppImage and also affects the graphical display.

Any help is appreciated!

AlbrechtL, 2019-10-26:
Thanks for reporting.
Please try the latest nightly build from https://bintray.com/albrechtl/welle.io/welle.io_nightly#files.

ontheair81, 2019-10-26:
Thank you! The latest nightly build (20191026_c85a9f8) fixed this problem.

Just for your information, I have seen two other issues with this nightly build:
- When scanning for stations there is no visible progress, it looks like welle.io is not responding anymore (which is not true).
- Station overview with PAD texts are not shown anymore (nothing happens when trying to add this sheet by activating it with the + button in the lower right corner).

tenzap, 2019-10-26:
Thanks for mentioning the display issues. Should be fixed once PR #455 is merged.

tenzap, 2019-11-14:
> Just for your information, I have seen two other issues with this nightly build:
>
>     * When scanning for stations there is no visible progress, it looks like welle.io is not responding anymore [...]

@ontheair81 , can you confirm the display issues you mentioned are fixed in the latest nightly ?

ontheair81, 2019-11-14:
> @ontheair81 , can you confirm the display issues you mentioned are fixed in the latest nightly ?

20191113_e942124_Linux_welle-io-x86_64.AppImage:
There is just the minimized icon in the task bar, but the program itself is not visible. "Maximize" or "Restore" does not work, Welle.io continues to be unvisible and there is no audio playback.

Tested on two PCs, both running Linux Mint 19.2, same behavior, sorry.

tenzap, 2019-11-14:
The version "20191113_e942124"  doesn't contain the fix, could you retry with latest which is https://bintray.com/albrechtl/welle.io/download_file?file_path=20191113_c3dc6be_Linux_welle-io-x86_64.AppImage

ontheair81, 2019-11-15:
@tenzap Yes, it is working! Scanning progress is shown and service overview is visible. Thank you!

AlbrechtL, 2019-11-24:
@ontheair81  Thanks for testing and reporting.
@tenzap  Thanks for fixing.

closed 2019-11-24 by AlbrechtL