AlbrechtL/welle.io #457: Software freezes on "Start station scan"

m0ukd, 2019-10-27:
I have an issue with version 2.0 where the software freezes when I click "Start station scan". It is only with version 2.0.

Version 2.0-Beta 3 was working fine. I've tried a few re-installs but no luck. I can only use 2.0-Beta 3.

Any ideas what this could be? I'm using standard RTL-SDR.

Thanks, John.

tenzap, 2019-10-27:
Likely same a #436, can you try by reverting commit e334317 on 2.0 release ?

tenzap, 2019-11-14:
Can you retry with latest nightly build ? It should be fixed since #460 has been merged

m0ukd, 2019-11-14:
Hi tenzap. Thanks for your help. I installed
20191113_ef3417d_Windows_welle-io-setup.exe
But I get an error when launching app of: Entry Point Not Found.
[image]

tenzap, 2019-11-15:
It seems to be a windows specific issue could you please file a new bug report with it ?

LA5HJA, 2019-11-17:
I did also try the Welle.io 2.0 and it did also freeze up on my win 10 pro N, when installing the Wello.io beta3 did it work wery well.

AlbrechtL, 2019-11-24:
>  But I get an error when launching app of: Entry Point Not Found.

I fixed the Windows issue in https://github.com/AlbrechtL/welle.io/commit/fd4e94885622708051af0d4ad8d990b69e3736cc. 

Please try again the latest nightly build.

closed 2019-12-02 by AlbrechtL