AlbrechtL/welle.io #459: AppImage 2.0 not working on Debian, same hardware on Windows/Android working :(

retrorich75, 2019-10-28:
Trying to use welle.io ver 2.0 with the following hardware : NooElec, NESDR Nano 3 under Debian 10 64 Bit and rtl_tcp results in no lock on signal.

Same Hardware under Windows 10 64 Bit and Android 9.0 phone or tablet works!

Not an expert the only thing different I see is the rtl_tcp programme under Linux is always changing the "gain".

rtl_tcp has this error in the output

"r82xx_write: i2c wr failed=-7 reg=05 len=1"

[image]

[image]

[image]

AlbrechtL, 2019-11-24:
Please post your expert settings.

AlbrechtL, 2019-12-06:
Please test it with version 2.1 again.

alexmyczko, 2019-12-18:
Here's a buster simple sid backport `http://phd-sid.ethz.ch/debian/welle.io/buster/` maybe you want to try that?

AlbrechtL, 2020-07-12:
Closed because of inactivity.

closed 2020-07-12 by AlbrechtL