AlbrechtL/welle.io #464: Unable to decode DAB streams with welle-cli version 2.0

beermad, 2019-11-10:
I have been using welle-cli for some time to timeshift DAB programs and it has become indispensable.
Unfortunately, since the arrival of version 2.0 (via the Arch Linux AUR) I have been unable to record from the web interface as there is no output.

How to reproduce:
1. Run welle-cli -c 12B -w 9876
2. Navigate browser to localhost:9876
3. Click on "play" for any channel within the MUX
With working versions, the audio plays. With 2.0, it doesn't. 
Similarly, trying to access the audio stream using streamripper works with older versions but no audio stream can be read with 2.0.
There are no error messages, however with a working version, as I click on "play", the program says:
Registering mp3 sender
MP2Decoder: using decoder 'AVX'.

With the failing version, the second line fails to appear. In addition, working versions can be terminated with <ctrl><c>, whereas this does not work with failing versions, which have to be killed from another terminal.

I have built various versions from git, with the following results:
v2.0-beta1: works
v2.0-beta2: works
v2.0-beta3: works
v2.0: fails.

Looking at the source, the only difference between beta3 and 2.0 in src/welle-cli is in webradiointerface.cpp and webradiointerface.h. Copying these from betat3 into 2.0 and rebuilding  results in a working binary, so I conclude it must be a change in one (or both) of these files which causes the problem. Unfortunately it is many years since I've done any C development, so investigating this further is not something I'm able to do.

I'm happy to try any suggestions to help further diagnose this.

mpbraendli, 2019-11-20:
Thanks for the report. There was indeed a regression in the changes you saw.
The `next` branch now includes commit 62e16be that should fix the mp3 stream again.

You might want to modify the PKGBUILD so that it compiles `next` and not `master`, not sure how to do this.
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=welle.io-git

Let me know if that solves the issue.

beermad, 2019-11-20:
Thanks. I can confirm that this fix works.
I'm building Welle after running: git clone -b next https://github.com/AlbrechtL/welle.io.git
I'll worry about building via the Arch AUR at a later date.

mpbraendli, 2019-11-21:
Yes this could be improved, we should cut outgoing connections on Ctrl-C too.

closed 2019-11-21 by mpbraendli