AlbrechtL/welle.io #466: RTL-SDR bad reception

Rootinator, 2019-11-13:
Hello,

I use welle.io with a RTL2832U (RTL-SDR) USB stick and have in some places very bad or no reception at all. With a DAB+ radio I have great reception in these places. 

I use an active DAB+ antenna.  

What could be the problem?

@AlbrechtL

andimik, 2019-11-13:
You must not compare a radio (hardware) with a software solution.

Does the radio play next to the PC?

Nevertheless, this seems to be a general question, not really tool related. You only would have good cards if other tools on the same PC play, on the other hand. 

You also don't mention which OS is used and which stick.

Rootinator, 2019-11-13:
I have connected the RTL SDR via a USB extension cable to prevent interference from the Raspberry Pi (it is using a Raspberry 3b+).  

The bad reception is also available without the USB extension cable. 

The operating system is Raspbian.

 The following stick is used: DVB-T + DAB + FM + RTL2832U + R820T2 (https://www.amazon.de/gp/aw/d/B013Q94CT6/)

FartherOut, 2019-11-23:
I had the same problem with a Nooelec stick. It worked OK for a while but then suddenly I could hardly get any reception of anything at all, even FM radio was poorly received.
It turns out I had managed to fry the chip with ESD, most likely by short circuiting the antenna ground and output, because the stick has a built-in bias-tee for powering antenna amplifiers.
The stick works... like normal... but simply can not receive anything strongly.

A general RTL stick without bias-tee should be harder to destroy like this, but shouldn't be ruled out either.
A simple static shock from your hands can destroy the stick, and it is always difficult to figure out if it has really happened, since the stick would seem to work, just not... properly.

AlbrechtL, 2019-11-28:
@FartherOut Thanks for the explanation.

closed 2019-11-28 by AlbrechtL