AlbrechtL/welle.io #469: licensing question (welle.io on macports & openssl)

tenzap, 2019-11-16:
Hello,

I've created the Portfile for macports.
Right now the binaries are not distributed, only the source code, so when someone installs welle.io through macports, compilation is done on the user's side instead of getting the binaries compiled by macports.

This is likely due to the fact that welle.io uses qtbase which uses openssl.
`"welle.io" is not distributable because its license "GPL-2" conflicts with license "OpenSSLException" of dependency "qt5-qtbase"`

I could add this line to the macports but I don't know if I'm allowed to do this and don't really understand it. What is you position on that ?
`license_noconflict      openssl`

The [doc on macports](https://guide.macports.org/#reference.keywords) states this :  
> license_noconflict
>     By default, it is assumed that ports may use libraries or headers from their dependencies and thus form a derivative work. A dependency with an incompatible license thus prevents the port from being distributed in binary form. If a dependency with an incompatible license is not used in such a way that a derivative work is formed, or should not prevent binary distribution for any other reason, add its name to this list.
> 
>     license_noconflict openssl

AlbrechtL, 2019-11-24:
Licensing of open source software and combining different open source licenses together is a really complicated thing. If you ask 10 different lawyers you will get 10 different answers.

The fact that Qt uses OpenSSL and these binaries are commonly used by many programs I think we can ignore the warning message.

tenzap, 2019-12-01:
FYI From : https://trac.macports.org/ticket/59779#comment:2

> [...] there is a license conflict here. The license in qt5-qtbase is {LGPL-3 GPL-3 OpenSSLException}. The license in welle.io is GPL-2. If these are both correct, then welle.io can't be distributed because GPLv3 and LGPLv3 have requirements that aren't in GPLv2 and are thus incompatible.

AlbrechtL, 2019-12-02:
From my point of view the code is under GPLv2 "or any later version".
See for example `welle.io/src/backend/dab-constants.cpp`
https://github.com/AlbrechtL/welle.io/blob/1afb20db95333d2b3dcf37e845962e4af36ca3d7/src/backend/dab-constants.cpp#L13-L16

tenzap, 2019-12-03:
Since there is some GPL-3 or any later (in backends) it should be licensed as GPL-3+ to distribute binaries in macports according to this [comment](https://trac.macports.org/ticket/59779#comment:7)

The files are licensed this way:
* backend/ : GPL-2+ GPL-3+
* input/ : GPL-2+
* tests/ : GPL-2+
* welle-cli/ : GPL-2+ (nothing specified in .js & .html)
* welle-gui/ : GPL-2+
  * waterfallitem.* : MIT
  * QML : no license specified

AlbrechtL, 2019-12-06:
So what we have to do? Exchanging https://github.com/AlbrechtL/welle.io/blob/d76a3b310a895c731f2ce3db94b800b5253cf9c2/COPYING by GPL-3?

tenzap, 2019-12-06:
Not sure what has to be done.
But would you agree that I put GPL-3+ (whih means GPL-3 or any later version) in welle.io macports portfile so that binary packages get distributed ?

AlbrechtL, 2019-12-06:
Please ask the guys from macports what has to be done.

tenzap, 2019-12-06:
May I suggest you to have a look at the full thread and maybe jump in ?

https://trac.macports.org/ticket/59779

Last comment says this 
> If there's any GPL-3+ in there, then the only way to comply with all the licenses when distributing binaries is to follow the conditions of GPLv3. So effectively it's just GPL-3+. 

I personally cannot decide anything concerning welle.io, that's up to you, sorry.

AlbrechtL, 2019-12-07:
From my point of view it is OK to got with GPL-3+ because the licenses says that also later GPL licenses are OK.

AlbrechtL, 2020-01-03:
Can we close this issue? As far as I see macports accepted welle.io.

closed 2020-04-30 by AlbrechtL