AlbrechtL/welle.io #472: SoapySDR : doesn't get the gain value at startup

tenzap, 2019-11-21:
This is with SoapySDR configured with rtl-sdr device and a manual gain.

In the GUI, at startup, the value of the gain is not updated and remains at zero, although the value seems correctly applied on the device. Moreover the slider position is ok, it is just the value displayed in the global settings page which is incorrect (the one written above the slider, and aligned right : `Value: x.xx`)

This values comes from file `soapy_sdr.cpp` and function setGain.
The problem is that "m_device->getGain(SOAPY_SDR_RX, 0);" return 0 while it should return the correct value.

BTW: 
- when moving the slider manually, the value is then correctly displayed, ie correctly returned by getGain, the problem is only at startup.
- if in setGain I repeat this call a 2nd time, then the gain is correctly displayed
`m_device->setGain(SOAPY_SDR_RX, 0, m_gains.at(gainIndex));` 
- There is no issue when using the device as 'native' "rtl-sdr" instead of through soapy-sdr.

AlbrechtL, 2019-11-24:
Thanks for reporting.

mpbraendli, 2020-02-04:
Maybe the RTLSDR needs some time before reading back the gain actually works. On the LimeSDR I'm unable to observe this.

I would propose we close this without fix, as the RTLSDR is better supported directly.

tenzap, 2020-02-04:
I understand this is not considered a critical issue, however, there are some cases where one wants to use Soapy with an RTLSDR device. In other words, using librtlsdr is not always better.

For example:
- on an older mac, the rtlsdr is unusable with RTLSDR. See #436. However when doing it through SoapySDR by setting manually a gain value it is fine.
- one may want to use SoapySDR in case one has more than 1 RTLSDR device since it permits to select the device to use through the soapy parameters (see #95 & my comment there https://github.com/AlbrechtL/welle.io/issues/95#issuecomment-562934690)

AlbrechtL, 2020-02-14:
I agree with @mpbraendli. We can reopen this issue any time.

closed 2020-02-14 by AlbrechtL