AlbrechtL/welle.io #476: [Windows] Texture atlas allocation failed, out of memory

mike-lt, 2019-12-05:
Hi, 

I'm using QT and I want to open the project on Windows. I downloaded the project (welle.io) and the required libraries (welle.io-win-libs) and put both of them on the same folder. 
I successfully built welle-gui without any errors but when I run it the program crashes and the Application Output gives me these errors. 

https://paste.ofcode.org/fdCiEHbQYrGyRJcuf9xEgC

Does anybody have any idea about the errors?

AlbrechtL, 2019-12-05:
The following lines from your log are showing that the issue is OpenGL related.
```
2019-12-05T16:36:07.008 Debug: QOpenGLFramebufferObject: Unsupported framebuffer format.
2019-12-05T16:36:07.008 Debug: QOpenGLFramebufferObject: Framebuffer incomplete, missing attachment.
2019-12-05T16:36:07.041 Debug: QSGTextureAtlas: texture atlas allocation failed, out of memory
16:36:09: The program has unexpectedly finished.
16:36:09: The process was ended forcefully.
```
Maybe it is an issue with your graphic driver.

mike-lt, 2019-12-06:
> The following lines from your log are showing that the issue is OpenGL related.
>
> ```
> 2019-12-05T16:36:07.008 Debug: QOpenGLFramebufferObject: Unsupported framebuffer format.
> 2019-12-05T16:36:07.008 [...]

Hmm, if there is an issue with the graphic card driver it shouldn't work in release mode as well but it works.
I have a laptop with gtx 1060 anyway I'll try to update the graphic card driver.

mike-lt, 2019-12-06:
I updated my driver but I'm stil having the same problem. I'll just gonna edit the program and build it in release mode.
Anyway, I am using this receiver: 
https://www.amazon.it/Portatile-Ricevente-Sintonizzatore-RTL2832U-Sostegno/dp/B013Q94CT6/ref=sr_1_5?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=dab+receiver&qid=1575640406&sr=8-5

The signal is very very low and I have to scan many times to get a list of radio channels. Then once it found a list it's very difficult to play a radio channel. 
During the scan operation I often get this error "ofdm-processor: SyncOnPhase failed". What does it specifically mean?

Thanks

AlbrechtL, 2019-12-06:
Which Qt version do you are using?

The message `ofdm-processor: SyncOnPhase failed` just means that the demodulator couldn't find a signal.

mike-lt, 2019-12-09:
I installed QT 5.12.6 and I'm building the project with "Desktop Qt 5.12.6 MinWG 32bit"

mike-lt, 2019-12-09:
I also have another problem. I edit the program by integrating a client socket.
Another program I created sent different messages to this program. I defined the following messages:
-start_scan
-stop_scan
-play [station_name]
-stop_radio

In radiocontroller.cpp I found all the methods I needed. I have a problem with the method CRadioController::stop() because it also "unplug" the RTL device (a message pops saying that there is no device) so once I call that method I can't call the play method anymore.

I looked at the stop method and this line code should be the culprit:
device->stop();

By removing it nothing happens. Then I also tried to add this line of code
audio->stop() 
The radio stops for a moment then it resumes.

Can anybody help me?
Thanks

AlbrechtL, 2019-12-09:
Do you mean that you are developing a remote control for welle.io? If yes, gerat!
To discuss it open a new issue please.

mike-lt, 2019-12-11:
Well, it's not really a remote control. It's just another program with other functionalities then I just added the possibility to send commands to welleio.

closed 2019-12-17 by mike-lt