AlbrechtL/welle.io #479: [rtl-sdr] closing welle.io after unplugging the device leads to crash

tenzap, 2019-12-07:
With 2.1 (+ commit d76a3b3) on linux
Steps:
- open welle.io
- play a station
- unplug rtlsdr device from computer
- then close welle.io

At closure time there is this crash.
```
Thread 1 (Thread 0x7f8e5f1e82c0 (LWP 18127)):
[KCrash Handler]
#6  0x00007f8e6373f7bb in raise () from /lib/x86_64-linux-gnu/libc.so.6
#7  0x00007f8e6372a535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#8  0x00007f8e63af4983 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007f8e63afa8c6 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007f8e63afa901 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#11 0x000056086cc304a6 in std::thread::~thread (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/8/thread:140
#12 CRTL_SDR::~CRTL_SDR (this=0x56086df8a060, __in_chrg=<optimized out>) at ./src/input/rtl_sdr.cpp:109
#13 0x000056086cc304b9 in CRTL_SDR::~CRTL_SDR (this=0x56086df8a060, __in_chrg=<optimized out>) at ./src/input/rtl_sdr.cpp:109
#14 0x000056086cc000a1 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x56086e0061c0) at /usr/include/c++/8/bits/shared_ptr_base.h:148
#15 std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x56086e0061c0) at /usr/include/c++/8/bits/shared_ptr_base.h:148
#16 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7ffd6f796e58, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/shared_ptr_base.h:728
#17 std::__shared_ptr<CVirtualInput, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7ffd6f796e50, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/shared_ptr_base.h:1167
#18 std::shared_ptr<CVirtualInput>::~shared_ptr (this=0x7ffd6f796e50, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/shared_ptr.h:103
#19 CRadioController::~CRadioController (this=0x7ffd6f796e30, __in_chrg=<optimized out>) at ./src/welle-gui/radio_controller.cpp:110
#20 0x000056086cbea4ef in main (argc=<optimized out>, argv=<optimized out>) at ./src/welle-gui/main.cpp:179
[Inferior 1 (process 18127) detached]
```
[welle-io-20191207-124000.kcrash.txt](https://github.com/AlbrechtL/welle.io/files/3935024/welle-io-20191207-124000.kcrash.txt)

Maybe related to #422 ? But the crash log is not the same.

AlbrechtL, 2019-12-07:
Thanks for reporting.

tenzap, 2019-12-08:
Since commit 37d6b10 error is now this way (line `#19` is new)
```
Thread 1 (Thread 0x7f116032a2c0 (LWP 22242)):
[KCrash Handler]
#6  0x00007f11648817bb in raise () from /lib/x86_64-linux-gnu/libc.so.6
#7  0x00007f116486c535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#8  0x00007f1164c36983 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007f1164c3c8c6 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007f1164c3c901 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#11 0x00005579de83d4d6 in std::thread::~thread (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/8/thread:140
#12 CRTL_SDR::~CRTL_SDR (this=0x5579e0bc27f0, __in_chrg=<optimized out>) at ./src/input/rtl_sdr.cpp:109
#13 0x00005579de83d4e9 in CRTL_SDR::~CRTL_SDR (this=0x5579e0bc27f0, __in_chrg=<optimized out>) at ./src/input/rtl_sdr.cpp:109
#14 0x00005579de80be4b in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x5579e0d3e7b0) at /usr/include/c++/8/bits/shared_ptr_base.h:148
#15 std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x5579e0d3e7b0) at /usr/include/c++/8/bits/shared_ptr_base.h:148
#16 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/shared_ptr_base.h:728
#17 std::__shared_ptr<CVirtualInput, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/shared_ptr_base.h:1167
#18 std::__shared_ptr<CVirtualInput, (__gnu_cxx::_Lock_policy)2>::reset (this=0x7ffcdf9dfbf0) at /usr/include/c++/8/bits/shared_ptr_base.h:1285
#19 CRadioController::closeDevice (this=0x7ffcdf9dfbd0) at ./src/welle-gui/radio_controller.cpp:121
#20 0x00005579de80cabe in CRadioController::~CRadioController (this=0x7ffcdf9dfbd0, __in_chrg=<optimized out>) at ./src/welle-gui/radio_controller.cpp:112
#21 0x00005579de7f74ef in main (argc=<optimized out>, argv=<optimized out>) at ./src/welle-gui/main.cpp:179
[Inferior 1 (process 22242) detached]
```
[welle-io-20191208-090256.kcrash.txt](https://github.com/AlbrechtL/welle.io/files/3935981/welle-io-20191208-090256.kcrash.txt)

AlbrechtL, 2019-12-09:
Fixed in https://github.com/AlbrechtL/welle.io/commit/2306088dda09b533691954932928eb0b854bae7d. Please verify it.

tenzap, 2019-12-09:
It fixes the crash, thank you !

However I noticed this:
- unplug the device and then close welle.io -> the log says `Reattaching kernel driver failed!`
```
2019-12-09T20:27:47.175 Debug: GUI: Close application
2019-12-09T20:27:47.176 Debug: main: Application closed
2019-12-09T20:27:47.201 Debug: GUI: Deleting CGUIHelper
2019-12-09T20:27:47.201 Debug: RadioController: Close device
2019-12-09T20:27:47.201 Info: OFDM-decoder:closing down now
Reattaching kernel driver failed!
2019-12-09T20:27:47.202 Debug: RadioController: Deleting CRadioController
2019-12-09T20:27:47.202 Warning: QObject::killTimer: Timers cannot be stopped from another thread
2019-12-09T20:27:47.220 Warning: QObject::~QObject: Timers cannot be stopped from another thread
2019-12-09T20:27:47.232 Warning: QFileSystemWatcher::removePaths: list is empty
2019-12-09T20:27:47.232 Warning: QFileSystemWatcher::removePaths: list is empty
```
- close welle.io but without unplugging the device -> the log says `Reattached kernel driver`
```
2019-12-09T20:32:35.729 Debug: GUI: Close application
2019-12-09T20:32:35.730 Debug: main: Application closed
2019-12-09T20:32:35.743 Debug: GUI: Deleting CGUIHelper
2019-12-09T20:32:35.743 Debug: RadioController: Close device
2019-12-09T20:32:35.743 Info: OFDM-processor: closing down
2019-12-09T20:32:35.743 Info: OFDM-decoder:closing down now
2019-12-09T20:32:36.792 Info: RTL_SDR: End rtlsdr_read_async_wrapper() thread
Reattached kernel driver
2019-12-09T20:32:36.995 Debug: RadioController: Deleting CRadioController
2019-12-09T20:32:36.996 Warning: QObject::killTimer: Timers cannot be stopped from another thread
2019-12-09T20:32:36.997 Warning: QObject::~QObject: Timers cannot be stopped from another thread
2019-12-09T20:32:37.011 Warning: QFileSystemWatcher::removePaths: list is empty
2019-12-09T20:32:37.012 Warning: QFileSystemWatcher::removePaths: list is empty
```

I don't know if this is expected behavior

AlbrechtL, 2019-12-09:
> `Reattaching kernel driver failed!`

This is coming from the librtlsdr and means that the dvb kernel modules cannot find the dongle. That is correct because you unplug it.

closed 2019-12-09 by AlbrechtL