AlbrechtL/welle.io #480: welle-cli segfaults when built with musl in Alpine Linux

probonopd, 2019-12-09:
#############################################
# Download and extract minimal Alpine system
#############################################

wget http://dl-cdn.alpinelinux.org/alpine/edge/releases/x86_64/alpine-minirootfs-3.11_rc1-x86_64.tar.gz
mkdir -p ./miniroot
cd ./miniroot
tar xf ../alpine-minirootfs-*.tar.gz
cd -

#############################################
# Prepare chroot
#############################################

sudo mount -o bind /dev miniroot/dev
sudo mount -t proc none miniroot/proc
sudo mount -t sysfs none miniroot/sys
sudo cp -p /etc/resolv.conf miniroot/etc/
sudo chroot miniroot /bin/sh <<\EOF

#############################################
# Now inside chroot
#############################################

apk add faad2-dev mpg123-dev alsa-lib-dev fftw-dev lame-dev  wget g++ make squashfs-tools desktop-file-utils curl appstream libc6-compat cmake git librtlsdr-dev --update-cache --repository http://dl-3.alpinelinux.org/alpine/edge/testing/ --allow-untrusted

git clone https://github.com/AlbrechtL/welle.io
cd welle.io/

mkdir build
cd build

cmake .. -DRTLSDR=1 -DBUILD_WELLE_IO=OFF

/welle.io/build # ./welle-cli 
Hello this is welle-cli v2.1-0-g8fc5739
InputFactory:Input device:auto
RTL_SDR:Open rtl-sdr
RTL_SDR:Found 1 devices. Uses the first working one
Found Rafael Micro R820T tuner
RTL_SDR: Opening rtl-sdr device0
[R82XX] PLL not locked!
RTL_SDR: Supported gain values29
RTL_SDR: gain 49.6
RTL_SDR: gain 48
RTL_SDR: gain 44.5
RTL_SDR: gain 43.9
RTL_SDR: gain 43.4
RTL_SDR: gain 42.1
RTL_SDR: gain 40.2
RTL_SDR: gain 38.6
RTL_SDR: gain 37.2
RTL_SDR: gain 36.4
RTL_SDR: gain 33.8
RTL_SDR: gain 32.8
RTL_SDR: gain 29.7
RTL_SDR: gain 28
RTL_SDR: gain 25.4
RTL_SDR: gain 22.9
RTL_SDR: gain 20.7
RTL_SDR: gain 19.7
RTL_SDR: gain 16.6
RTL_SDR: gain 15.7
RTL_SDR: gain 14.4
RTL_SDR: gain 12.5
RTL_SDR: gain 8.7
RTL_SDR: gain 7.7
RTL_SDR: gain 3.7
RTL_SDR: gain 2.7
RTL_SDR: gain 1.4
RTL_SDR: gain 0.9
RTL_SDR: gain 0
Start RTLSDR thread
OFDM-processor: reset (0)
OFDM-processor:start
Wait for sync
Segmentation fault

#############################################
# Exit chroot and clean up
#############################################

exit
EOF
sudo umount miniroot/proc miniroot/sys miniroot/dev

mpbraendli, 2019-12-11:
Hi, can you make a debug build and give me a backtrace of the segfault?

probonopd, 2019-12-11:
I can give it a try if you can tell me how to do that...

mpbraendli, 2019-12-12:
For the debug build, use the `-DCMAKE_BUILD_TYPE=Debug` option when calling cmake.

For getting a backtrace, do `gdb --args ./welle-cli`, and then type `run` in the gdb command line interface. After the segfault it will drop you back to the gdb ui. Type `bt full` to get the backtrace.

gvanem, 2020-02-01:
Maybe my issue is also related to this:
Running from the cmd-line, a `welle-cli` could suddenly stop while serving a web-client.
But while running inside the Windows Console debugger, I see this exception gets caught:
```
ofdm-processor: SyncOnPhase failed
(f28.2b28): C++ EH exception - code e06d7363 (first chance)
...
004fe828 5f9b27ce MSVCP140!__ExceptionPtrRethrow(void * _PtrRaw = 0x004fe858)+0xb3
004fe850 5f9b2782 MSVCP140!std::rethrow_exception(class std::exception_ptr _Ptr = class std::exception_ptr)+0x19
004fe880 00a56126 MSVCP140!std::_Rethrow_future_exception(class std::exception_ptr _Ptr = class std::exception_ptr)+0x22
004fe8c0 00a56232 welle_cli!std::_Associated_state<bool>::_Get_value(bool _Get_only_once = true)+0x86
004fe8d0 00a561ff welle_cli!std::_Task_async_state<bool>::_Get_value(bool _Get_only_once = true)+0x22
004fe8f0 00a64002 welle_cli!std::_State_manager<bool>::_Get_value(void)+0x4f
004fe914 00a6f5fa welle_cli!std::future<bool>::get(void)+0x42
004fe9a0 00a79805 welle_cli!WebRadioInterface::serve(void)+0x16a
004ff9d0 00b5ee96 welle_cli!main(int argc = 0n7, char ** argv = 0x006e3740)+0x7d5
```

I've no idea what that `deque<future<bool> >` in `webradiointerface.cpp` is and why my MSVC is bitching on it.

mpbraendli, 2020-02-04:
@gvanem your observation might be another issue, I'm not sure. Is this the complete log or did you cut off a part of it?

@probonopb Can you try again with latest "next"? commit 635dd53 fixes an issue about unaligned memory access that might trigger a segfault on Alpine since that uses another libc (If I'm not mistaken)

probonopd, 2020-02-04:
I can launch it and I can start to use the web interface; however after switching channels a few time I still get a segfault in `next`.

```
OFDM-processor:restart
RETUNE Start programme handler
POST channel: 11B
RETUNE: Retune to 218640000
TEARDOWN Cancel all PHs and remove services
TEARDOWN Stop rx
OFDM-processor: closing down
RETUNE Take ownership of RX
RETUNE Destroy RX
OFDM-decoder:closing down now
RETUNE Set frequency
RTL_SDR: End rtlsdr_read_async_wrapper() thread
RETUNE Restart RX
RTL_SDR: Start rtlsdr_read_async_wrapper() thread
OFDM-processor:restart
RETUNE Start programme handler
POST channel: 7B
RETUNE: Retune to 190640000
TEARDOWN Cancel all PHs and remove services
TEARDOWN Stop rx
OFDM-processor: closing down
RETUNE Take ownership of RX
RETUNE Destroy RX
OFDM-decoder:closing down now
RETUNE Set frequency
RTL_SDR: End rtlsdr_read_async_wrapper() thread
RETUNE Restart RX
RTL_SDR: Start rtlsdr_read_async_wrapper() thread
OFDM-processor:restart
RETUNE Start programme handler
Segmentation fault
```

probonopd, 2020-02-04:
`./welle-cli -c 10A -w 7979 -C 2` does not segfault, whereas `./welle-cli -c 11A -w 7979 -C 2` does segfault. On 11A there is SWR3 (which I can receive with the welle.io official AppImage)

mpbraendli, 2020-02-05:
That seems like a different issue then :-/
Can you capture an IQ file that leads to a segfault and share it with us? I'll setup a chroot with alpine linux inside to try to reproduce that.

mpbraendli, 2020-02-05:
I'm able to reproduce the issue with an IQ file I had already.
```
# gdb --args ./welle-cli -f ./short.u8.iq -w 7979 -D
[...]
(gdb) run
[...]
Thread 5 "welle-cli" received signal SIGSEGV, Segmentation fault.
[...]
(gdb) bt
#0  0x00005555555fd9c0 in RingBuffer<unsigned char>::getDataFromBuffer (elementCount=-282867632, data=0x7ffff6322810, this=0x7ffff791e428) at /usr/include/c++/9.2.0/bits/stl_vector.h:1040
#1  CRAWFile::convertSamples (this=this@entry=0x7ffff791e3e0, Buffer=..., V=V@entry=0x7ffff61a6020, size=size@entry=2048) at /welle.io/src/input/raw_file.cpp:331
#2  0x00005555555fe9e3 in CRAWFile::getSamples (size=<optimized out>, V=<optimized out>, this=<optimized out>) at /welle.io/src/input/raw_file.cpp:224
#3  CRAWFile::getSamples (this=0x7ffff791e3e0, V=0x7ffff61a6020, size=2048) at /welle.io/src/input/raw_file.cpp:213
#4  0x00005555555edb4f in OFDMProcessor::getSamples (this=0x555555669760, v=0x7ffff61a6020, n=2048, phase=0) at /welle.io/src/backend/ofdm-processor.cpp:208
#5  0x00005555555ef041 in OFDMProcessor::run (this=0x555555669760) at /welle.io/src/backend/ofdm-processor.cpp:339
#6  0x00007ffff74db045 in ?? () from /usr/lib/libstdc++.so.6
#7  0x00007ffff7fba72b in ?? () from /lib/ld-musl-x86_64.so.1
#8  0x0000000000000000 in ?? ()
```

mpbraendli, 2020-02-05:
It seems musl uses a smaller default stack size than glibc. See "Thread stack size" on https://wiki.musl-libc.org/functional-differences-from-glibc.html
If I link with `-Wl,-z,stack-size=512000` it welle-cli works.

We could maybe call `pthread_attr_setstacksize` as suggested, but I don't know about other platforms.

```
cmake .. -DRTLSDR=1 -DBUILD_WELLE_IO=OFF -DCMAKE_EXE_LINKER_FLAGS=-Wl,-z,stack-size=512000
```

Please try again with this cmake invocation.

probonopd, 2020-02-10:
Thanks @mpbraendli, with this I get no more segfault:

```
/welle.io/build # ./welle-cli -c 11A -w 7979 -C 2
Hello this is welle-cli v2.1-62-gdb596b3
InputFactory:Input device:auto
RTL_SDR: Open rtl-sdr
RTL_SDR: Found 1 devices. Uses the first working one
Detached kernel driver
Found Rafael Micro R820T tuner
RTL_SDR:  Opening rtl-sdr device0
[R82XX] PLL not locked!
RTL_SDR:  Supported gain values29
RTL_SDR:  gain 49.6
RTL_SDR:  gain 48
RTL_SDR:  gain 44.5
RTL_SDR:  gain 43.9
RTL_SDR:  gain 43.4
RTL_SDR:  gain 42.1
RTL_SDR:  gain 40.2
RTL_SDR:  gain 38.6
RTL_SDR:  gain 37.2
RTL_SDR:  gain 36.4
RTL_SDR:  gain 33.8
RTL_SDR:  gain 32.8
RTL_SDR:  gain 29.7
RTL_SDR:  gain 28
RTL_SDR:  gain 25.4
RTL_SDR:  gain 22.9
RTL_SDR:  gain 20.7
RTL_SDR:  gain 19.7
RTL_SDR:  gain 16.6
RTL_SDR:  gain 15.7
RTL_SDR:  gain 14.4
RTL_SDR:  gain 12.5
RTL_SDR:  gain 8.7
RTL_SDR:  gain 7.7
RTL_SDR:  gain 3.7
RTL_SDR:  gain 2.7
RTL_SDR:  gain 1.4
RTL_SDR:  gain 0.9
RTL_SDR:  gain 0
RTL_SDR: Start rtlsdr_read_async_wrapper() thread
OFDM-processor:restart
ofdm-processor: Lost coarse sync (coarseCorrector: 0; fineCorrector: 0)
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
ofdm-processor: SyncOnPhase failed
```

I don't hear anything, but that may well be due to my chrooted Alpine Linux setup. So I'd say this fixes the original issue and this ticket can be closed.

closed 2020-02-10 by probonopd

mpbraendli, 2020-02-11:
Thanks for the confirmation.

I don't see a reason why you have no sound, because it says
`RTL_SDR: Found 1 devices. Uses the first working one`

Is the webpage accessible? Do you see a list of services?

andimik, 2020-02-11:
You could try on another machine or mobile phone using the IP Address of the welle-cli device. I am pretty sure that you will have sound there.

probonopd, 2020-02-13:
Yes, probably. Will investigate more. In any case, no more segfaults :+1: