AlbrechtL/welle.io #482: Add a Pause/Stop button

jpuigs, 2019-12-12:
It would be good to have a stop button, in order to stop playing instead of need to close program or app..

AlbrechtL, 2020-01-03:
Thanks for your idea!

tenzap, 2020-03-30:
see PR #532

Jasoroony, 2020-12-24:
I see all of this went no-where... but I'd like to see a stop button (detuned, like when you start the app) too.

andimik, 2020-12-24:
You could use his PR and build welle.io. For that you even need not fork it.

Jasoroony, 2020-12-24:
Thanks, yes I will try that.  However, I'm not at all experienced with GitHub and how all this stuff works.  I've saved the html files here that show the differences - so I can apply them by hand.  It seems his repository is outdated, so I don't want to lose the latest updates in this one.

andimik, 2020-12-24:
Then fork welle.io and make a PR of #532 to your fork.

Jasoroony, 2020-12-30:
@tenzap - thank-you!  I didn't copy over your start/stop mod, but I did implement the previous/next channel shortcuts (and using your awesome example code, a couple of my own shortcuts that I needed).  It's a shame you're so busy, but I understand, I'm pretty busy too with my own things ;)

Jasoroony, 2020-12-30:
Oh, and my "top bar" now says "DAB+ Radio" instead of "welle-io" - since I've had to make my own changes, I might as well "make it my own". :)

tenzap, 2021-02-27:
play/stop has been merged in next. This issue can be closed I guess

closed 2021-03-27 by AlbrechtL