AlbrechtL/welle.io #483: CMake/Ninja build is failing on Debian Sid

russel, 2019-12-17:
On Debian Sid, using g++ "gcc version 9.2.1 20191130 (Debian 9.2.1-21)", I am getting a build fail:
```
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/viterbi.cpp: In member function ‘void Viterbi::deconvolve(softbit_t*, uint8_t*)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/viterbi.cpp:234:18: warning: comparison is always false due to limited range of data type [-Wtype-limits]
  234 |         if (temp < 0) temp = 0;
      |             ~~~~~^~~
[21/76] Building CXX object CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o
FAILED: CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o 
/usr/bin/c++  -DDABLIN_AAC_FAAD2 -DGITDESCRIBE=\"v2.1-6-g2e6fab2\" -DGITHASH=\"2e6fab2\" -DHAVE_ALSA -DHAVE_SOAPYSDR -Iwelle-cli_autogen/include -I/home/users/russel/Repositories/Git/Forks/WelleIo/src -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/output -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/various -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/input -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/gui -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/libs/fec -O3 -DNDEBUG   -Wall -g -Wall -Wextra -std=gnu++14 -MD -MT CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o -MF CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o.d -o CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o -c /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:99:22: error: ‘virtual void AlsaProgrammeHandler::onMOT(const std::vector<unsigned char>&, int)’ marked ‘override’, but does not override
   99 |         virtual void onMOT(const std::vector<uint8_t>& data, int subtype) override { (void)data; (void)subtype; }
      |                      ^~~~~
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp: In function ‘int main(int, char**)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:551:34: error: cannot declare variable ‘ph’ to be of abstract type ‘AlsaProgrammeHandler’
  551 |             AlsaProgrammeHandler ph;
      |                                  ^~
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:72:7: note:   because the following virtual functions are pure within ‘AlsaProgrammeHandler’:
   72 | class AlsaProgrammeHandler: public ProgrammeHandlerInterface {
      |       ^~~~~~~~~~~~~~~~~~~~
In file included from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/virtual_input.h:38,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/soapy_sdr.h:31,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:45:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/radio-controller.h:165:22: note: 	‘virtual void ProgrammeHandlerInterface::onMOT(const mot_file_t&)’
  165 |         virtual void onMOT(const mot_file_t& mot_file) = 0;
      |                      ^~~~~
[30/76] Building CXX object CMakeFiles/welle-io.dir/welle-io_autogen/mocs_compilation.cpp.o
```

alexmyczko, 2019-12-18:
Also `gcc (Debian 9.2.1-21) 9.2.1 20191130` here, builds fine for me (2.1 of welle.io), do you use a later cloen?

russel, 2019-12-18:
@alexmyczko   I am building master/HEAD of WelleIO – obviously I should have said that. :-)

alexmyczko, 2019-12-18:
a ```
wget https://github.com/AlbrechtL/welle.io/archive/master.tar.gz
tar xzvf master.tar.gz
cd we*/
cmake .
make
```
worked for me successfully, how are you building?

```
[ 95%] Building CXX object CMakeFiles/welle-io.dir/src/input/null_device.cpp.o
[ 96%] Building CXX object CMakeFiles/welle-io.dir/src/input/raw_file.cpp.o
[ 97%] Building CXX object CMakeFiles/welle-io.dir/src/input/rtl_tcp.cpp.o
[ 98%] Building CXX object CMakeFiles/welle-io.dir/welle-io_autogen/MOX6SOXE6H/qrc_resources.cpp.o
[100%] Linking CXX executable welle-io
[100%] Built target welle-io
```

russel, 2019-12-18:
I use:
```
cmake -DCMAKE_INSTALL_PREFIX=$HOME/Built -DRTLSDR=YES -DSOAPYSDR=YES -G Ninja ~/Repositories/Git/Forks/WelleIo
```
in `~/BuildArea/WelleIO` to get the Ninja build, and then just run Ninja.
```
|> ninja
[0/1] Re-running CMake...
-- Build type not specified: defaulting to release.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/users/russel/BuildArea/WelleIO
[4/96] Building CXX object CMakeFiles/welle-cli.dir/src/backend/viterbi.cpp.o
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/viterbi.cpp: In member function ‘void Viterbi::deconvolve(softbit_t*, uint8_t*)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/viterbi.cpp:234:18: warning: comparison is always false due to limited range of data type [-Wtype-limits]
  234 |         if (temp < 0) temp = 0;
      |             ~~~~~^~~
[25/96] Building CXX object CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o
FAILED: CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o 
/usr/bin/c++  -DDABLIN_AAC_FAAD2 -DGITDESCRIBE=\"v2.1-17-g123f32a\" -DGITHASH=\"123f32a\" -DHAVE_ALSA -DHAVE_RTLSDR -DHAVE_SOAPYSDR -Iwelle-cli_autogen/include -I/home/users/russel/Repositories/Git/Forks/WelleIo/src -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/output -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/various -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/input -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/gui -I/home/users/russel/Repositories/Git/Forks/WelleIo/src/libs/fec -O3 -DNDEBUG   -Wall -g -Wall -Wextra -std=gnu++14 -MD -MT CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o -MF CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o.d -o CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o -c /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:99:22: error: ‘virtual void AlsaProgrammeHandler::onMOT(const std::vector<unsigned char>&, int)’ marked ‘override’, but does not override
   99 |         virtual void onMOT(const std::vector<uint8_t>& data, int subtype) override { (void)data; (void)subtype; }
      |                      ^~~~~
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp: In function ‘int main(int, char**)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:551:34: error: cannot declare variable ‘ph’ to be of abstract type ‘AlsaProgrammeHandler’
  551 |             AlsaProgrammeHandler ph;
      |                                  ^~
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:72:7: note:   because the following virtual functions are pure within ‘AlsaProgrammeHandler’:
   72 | class AlsaProgrammeHandler: public ProgrammeHandlerInterface {
      |       ^~~~~~~~~~~~~~~~~~~~
In file included from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/virtual_input.h:38,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/soapy_sdr.h:31,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:45:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/radio-controller.h:165:22: note: 	‘virtual void ProgrammeHandlerInterface::onMOT(const mot_file_t&)’
  165 |         virtual void onMOT(const mot_file_t& mot_file) = 0;
      |                      ^~~~~
[34/96] Building CXX object CMakeFiles/welle-cli.dir/src/welle-cli/webradiointerface.cpp.o
ninja: build stopped: subcommand failed.
```
If I avoid use of Ninja, and instead use:
```
cmake -DCMAKE_INSTALL_PREFIX=$HOME/Built -DRTLSDR=YES -DSOAPYSDR=YES ~/Repositories/Git/Forks/WelleIo
```
to get a Make build then:
```
|> make
Scanning dependencies of target welle-cli_autogen
[  1%] Automatic MOC for target welle-cli
[  1%] Built target welle-cli_autogen
Scanning dependencies of target welle-cli
[  2%] Building CXX object CMakeFiles/welle-cli.dir/welle-cli_autogen/mocs_compilation.cpp.o
[  3%] Building CXX object CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:99:22: error: ‘virtual void AlsaProgrammeHandler::onMOT(const std::vector<unsigned char>&, int)’ marked ‘override’, but does not override
   99 |         virtual void onMOT(const std::vector<uint8_t>& data, int subtype) override { (void)data; (void)subtype; }
      |                      ^~~~~
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp: In function ‘int main(int, char**)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:551:34: error: cannot declare variable ‘ph’ to be of abstract type ‘AlsaProgrammeHandler’
  551 |             AlsaProgrammeHandler ph;
      |                                  ^~
/home/users/russel/Repositories/Git/Forks/WelleIo/src/welle-cli/welle-cli.cpp:72:7: note:   because the following virtual functions are pure within ‘AlsaProgrammeHandler’:
   72 | class AlsaProgrammeHandler: public ProgrammeHandlerInterface {
      |       ^~~~~~~~~~~~~~~~~~~~
In file included from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/virtual_input.h:38,
                 from /home/users/russel/Repositories/Git/Forks/WelleIo/src/input/soapy_sdr.h:31,
[... cut ...]
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/radio-controller.h:165:22: note: 	‘virtual void ProgrammeHandlerInterface::onMOT(const mot_file_t&)’
  165 |         virtual void onMOT(const mot_file_t& mot_file) = 0;
      |                      ^~~~~
make[2]: *** [CMakeFiles/welle-cli.dir/build.make:76: CMakeFiles/welle-cli.dir/src/welle-cli/welle-cli.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/welle-cli.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
so there seems to be consistency.

russel, 2019-12-18:
I made one big error in a previous message: I am not using master/HEAD, I am using next/HEAD. The idea is not to build for deployment, but is to build to ensure with every change that the CMake build works.

AlbrechtL, 2019-12-21:
Should be fixed in https://github.com/AlbrechtL/welle.io/commit/0067b7b74c1a3677786ba71b0ddd943c94363bdd. Please feel free to verify it.

russel, 2019-12-21:
@AlbrechtL  I just refreshed to the latest next/MASTER and:
```
|> ninja
[23/96] Building CXX object CMakeFiles/welle-cli.dir/src/backend/viterbi.cpp.o
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/viterbi.cpp: In member function ‘void Viterbi::deconvolve(softbit_t*, uint8_t*)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/viterbi.cpp:234:18: warning: comparison is always false due to limited range of data type [-Wtype-limits]
  234 |         if (temp < 0) temp = 0;
      |             ~~~~~^~~
[71/96] Building CXX object CMakeFiles/welle-io.dir/src/backend/viterbi.cpp.o
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/viterbi.cpp: In member function ‘void Viterbi::deconvolve(softbit_t*, uint8_t*)’:
/home/users/russel/Repositories/Git/Forks/WelleIo/src/backend/viterbi.cpp:234:18: warning: comparison is always false due to limited range of data type [-Wtype-limits]
  234 |         if (temp < 0) temp = 0;
      |             ~~~~~^~~
[96/96] Linking CXX executable welle-cli
```
So yes, this issue is now fixed.

AlbrechtL, 2019-12-21:
Great! Thanks for testing.

closed 2019-12-21 by AlbrechtL