AlbrechtL/welle.io #487: Stations listed by channel groups

Observerist, 2019-12-23:
The v2.1 GUI is absolutely amazing. Got some free time to explore themes and modules ( Material + Dark is my Fave ). 

Is there a way to display a station list grouped by channels rather than just by alphabetic order ? Switchable with the present display, if possible.

Love welle.io now even more, since Macports takes care of everything without having to go through Qt Creator to compile it. That part of the building routine was always painful.

AlbrechtL, 2020-01-03:
Thanks for your idea!

Can you post a mockup how it can look like?

Observerist, 2020-01-04:
Using for the purpose of clarity just two muxes, Paris' 9D & 11A, that have different “loads”, so that the stations listed inside a mux are easy to identify as more or less populated.

There is only need for the channel nº of the mux and its tagged ID name.

If this sorting mode could be chosen in both the regular and expert mode, meaning the sorting selector would come on top of the mode selection, that would be ideal.
The filterd list of only favorites obviously doesn’t need that sorting choice.

Although the mux Identification line itself is written ‘justified left’, with 4 spaces between the ch nº and its name, the positioning of the whole line is offset from the left border as to make it easier to read and identify.

btw, I used the calibri font, which seems to be the closest to your choice.

Not a big mod but it makes sorting and selecting more flexible. Would this be useful ?

[image]

[image]

AlbrechtL, 2020-01-04:
[image]

I'm liking that idea pretty much! We will put that to our to do list.

Anyone is invited to implement it and to send a pull request.

starman316uk, 2020-04-19:
I'd love this implemented too

andimik, 2024-12-27:
AbracaDABra has an icon to switch between the same service if it is on different multiplexes.

[image]

[image]

AlbrechtL, 2024-12-27:
I added in https://github.com/AlbrechtL/welle.io/commit/ba8b330dbdb763c26ed96d1be08eb76c268e296c the initial support for stations that available in different ensembles.

[image]

The basic function should be there. Please test it and report if it is working or not.

Nice to have would be to show the ensemble name instead of the channel name.

andimik, 2024-12-28:
What data for duplicates is used? SId and Label or SId only?

AlbrechtL, 2024-12-28:
Only the SId.

AlbrechtL, 2025-03-05:
Is any feedback available?

andimik, 2025-03-07:
I found the correct flatpak version, but it was a bit tricky as it is not listed by date and time, so I needed to look for the correct commit. 

https://welle-io-nightlies.albrechtloh.de/20241227_c99d2184_Linux_welle-io-x86_64.flatpak with commit c99d2184 is the right one.

Yes, this works, I have some services in various muxes.

[image]

AlbrechtL, 2025-03-12:
In https://github.com/AlbrechtL/welle.io/commit/19870e0fac87d49ffbd4fa0060f8f1646541eb39 I added to display the ensemble name instead of the channel name. The channel name is only be used of no ensemble name was found.

[image]

@andimik It would be really nice if you could test it again. Because the ensemble names are not saves in previous welle.io versions, you need to tune to each ensemble to retrieve the name. Otherwise, the channel name is shown.

andimik, 2025-03-12:
Will do whenever the flatpack version is working, see https://github.com/AlbrechtL/welle.io/issues/850

AlbrechtL, 2025-03-13:
@andimik Please try this flatpak: https://welle-io-nightlies.albrechtloh.de/20250313_welle-io_group_station_test.flatpak. Currently, I don't know why the automatically created build are not working.

AlbrechtL, 2025-03-16:
Included in version https://github.com/AlbrechtL/welle.io/releases/tag/v2.7

closed 2025-03-16 by AlbrechtL