AlbrechtL/welle.io #492: welle-cli has incorrect help text

beermad, 2019-12-28:
Hi,
I've just installed 2.1 Git: 91319f3 from Arch's AUR, replacing v2.0-109-gb651721 which I'd installed from source.
I've noticed that welle-cli -h displays the options for welle.io, not for welle-cli. Checking back to v2.0-109-gb651721 I can see the same help there, so it seems this change occurred earlier.
This isn't a problem for me personally, as I have everything set up and working beautifully, but it could present problems for any new users wanting to understand how to use this excellent piece of code.

mpbraendli, 2020-01-03:
Hi,

first, thanks for the compliment! :-)

I'm however unable to understand how this could happen, and the binary built with cmake doesn't show this.

andimik, 2020-01-03:
I cannot confirm this issue as well and would suggest to close it.

```
$ welle-cli -h
Hello this is welle-cli v2.0-33-g5385057
Usage: 
Receive using RTLSDR, and play with ALSA:
 welle-cli -c channel -p programme

Read an IQ file and play with ALSA:
IQ file format is u8, unless the file ends with FORMAT.iq
 welle-cli -f file -p programme

Use -D to dump FIC and all programmes to files.
 welle-cli -c channel -D 

Use -w to enable webserver, decode a programmes on demand.
 welle-cli -c channel -w port

Use -Dw to enable webserver, decode all programmes.
 welle-cli -c channel -Dw port

Use -C 1 -w to enable webserver, decode programmes one by one in a carousel.
Use -C N -w to enable webserver, decode programmes N by N in a carousel.
This is useful if your machine cannot decode all programmes simultaneously, but
you still want to get an overview of the ensemble.
Without the -P option, welle-cli will switch every 10 seconds.
With the -P option, welle-cli will switch once DLS and a slide were decoded, staying at most
80 seconds on a given programme.
 welle-cli -c channel -C 1 -w port
 welle-cli -c channel -PC 1 -w port

Backend and input options
 -u      disable coarse corrector, for receivers who have a low frequency offset.
 -g GAIN set input gain to GAIN or -1 for auto gain.
 -A ANT  set input antenna to ANT (for SoapySDR input only).
 -T      disable TII decoding to reduce CPU usage.

Use -t test_number to run a test.
To understand what the tests do, please see source code.

 examples: welle-cli -c 10B -p GRRIF
           welle-cli -f ./ofdm.iq -p GRRIF
           welle-cli -f ./ofdm.iq -t 1

```

mpbraendli, 2020-02-04:
Please open the issue again if necessary.

closed 2020-02-04 by mpbraendli