AlbrechtL/welle.io #494: Problems compiling welle.io for Raspberry

Rootinator, 2019-12-30:
Hi there,

I would like to compile welle.io for the Raspberry, but there are problems.

I have successfully installed the QTCreator, but I cannot compile welle.io correctly.

According to the instructions, you have to comment out Airspy in the wave.io.pro, there is nothing from Airspy. You can only find this in "tests/backend/backend.pri" line 17. I commented it out.

Then after clicking on the hammer there are already some errors in the QTCreator:

`/home/a/Desktop/welle.io/../welle.io/src/backend/viterbi.cpp:234: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         if (temp < 0) temp = 0;
                  ^`

`/home/a/Desktop/welle.io/../welle.io/src/backend/viterbi.cpp:234: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         if (temp < 0) temp = 0;
                  ^`

`/home/a/Desktop/welle.io/../welle.io/src/backend/viterbi.cpp:234: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         if (temp < 0) temp = 0;
                  ^`

`/home/a/Desktop/welle.io/../welle.io/src/welle-cli/tests.cpp:368: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 8 has type ‘std::chrono::duration<long long int, std::ratio<1ll, 1000ll> >::rep {aka long long int}’ [-Wformat=]
             std::accumulate(tph.rsErrorStats.begin(), tph.rsErrorStats.end(), 0));
                                                                                 ^`

`/home/a/Desktop/welle.io/../welle.io/src/welle-gui/gui_helper.cpp:543: warning: unused parameter ‘fileFormat’ [-Wunused-parameter]
 void CGUIHelper::openRawFile(QString fileFormat)
                                      ^`

Actually, a file (or a folder?) Named "welle-io" should have been created. But this is not the case.
In the release folder "build-welle.io-Raspberry_Pi-Release" there is only a folder "src" and a file "Makefile".

How can I compile welle.io for Rasperry?

My system is as described in the instructions:
Ubuntu 17.10
QTCreator 5.9.3

@AlbrechtL

mpbraendli, 2019-12-30:
Hi, thanks for the report.
However, without the actual error message, there's no way for us to help you. Please provide the error that leads to the build failure.

Rootinator, 2019-12-30:
Unfortunately there are no other error messages than the one I added ...

I have put together several screenshots to show my steps:

[image]

gvanem, 2019-12-30:
So what happens when you do a `make` in the `src` folder?

andimik, 2019-12-30:
Just fyi; I've successfully compiled welle-cli on a RPI3B using `cmake`

Rootinator, 2019-12-30:
`make` in the src folder:

`a@ubuntu:~/Desktop/build-welle.io-Raspberry_Pi-Release/src$ make
make: *** No targets specified and no makefile found.  Stop.`

`make` in the build folder:

`a@ubuntu:~/Desktop/build-welle.io-Raspberry_Pi-Release$ make
cd src/welle-gui/ && ( test -e Makefile || /home/a/raspi/qt5/bin/qmake -o Makefile /home/a/Desktop/welle.io/src/welle-gui/welle-gui.pro -spec devices/linux-rasp-pi3-g++ ) && make -f Makefile 
make[1]: Entering directory '/home/a/Desktop/build-welle.io-Raspberry_Pi-Release/src/welle-gui'
make[1]: Nothing to be done for 'first'.
make[1]: Leaving directory '/home/a/Desktop/build-welle.io-Raspberry_Pi-Release/src/welle-gui'
cd src/welle-cli/ && ( test -e Makefile || /home/a/raspi/qt5/bin/qmake -o Makefile /home/a/Desktop/welle.io/src/welle-cli/welle-cli.pro -spec devices/linux-rasp-pi3-g++ ) && make -f Makefile 
make[1]: Entering directory '/home/a/Desktop/build-welle.io-Raspberry_Pi-Release/src/welle-cli'
make[1]: Nothing to be done for 'first'.
make[1]: Leaving directory '/home/a/Desktop/build-welle.io-Raspberry_Pi-Release/src/welle-cli'
cd src/tests/ && ( test -e Makefile || /home/a/raspi/qt5/bin/qmake -o Makefile /home/a/Desktop/welle.io/src/tests/tests.pro -spec devices/linux-rasp-pi3-g++ ) && make -f Makefile 
make[1]: Entering directory '/home/a/Desktop/build-welle.io-Raspberry_Pi-Release/src/tests'
make[1]: Nothing to be done for 'first'.
make[1]: Leaving directory '/home/a/Desktop/build-welle.io-Raspberry_Pi-Release/src/tests'`

Rootinator, 2019-12-31:
Do you already know where the error is and why I cannot compile welle.io?

AlbrechtL, 2020-01-03:
> However, I am not sure why welle.io is not compiled successfully ...

The compiler messages that you posted are warnings but there are no errors. Please post the last compiler output messages.

seife, 2020-01-04:
On a "naked" fresh installation of 2019-09-26-raspbian-buster-lite.img, this works for me:

```
sudo apt-get install cmake git
sudo apt-get install qt5-default qtquickcontrols2-5-dev qtdeclarative5-dev \
    qtmultimedia5-dev libqt5charts5-dev libmp3lame-dev libfftw3-dev \
    libmpg123-dev libsoapysdr-dev librtlsdr-dev libairspy-dev \
    libasound2-dev libfaad-dev

git clone https://github.com/AlbrechtL/welle.io.git
cd welle.io
cd build
cmake .. -DRTLSDR=1 -DAIRSPY=1 -DSOAPYSDR=1
make -j2
```

Rootinator, 2020-01-06:
After a complete reinstallation from the host, I was now able to compile welle.io without any problems.

I don't know what the error was, but now it worked.

AlbrechtL, 2020-01-15:
@seife Thanks for testing
@Rootinator Great to hear that the issue disappeared.

closed 2020-01-15 by AlbrechtL

andimik, 2020-02-02:
@Rootinator

Was it the same like in https://forum-raspberrypi.de/forum/thread/44257-fehler-beim-kompilieren-von-welle-io-qt/ ?