AlbrechtL/welle.io #496: backend: possible issue in Viterbi::deconvolve?

seife, 2020-01-04:
I noticed a warning from the compiler today:

```
.../welle.io/src/backend/viterbi.cpp: In member function ‘void Viterbi::deconvolve(softbit_t*, uint8_t*)’:
.../welle.io/src/backend/viterbi.cpp:234:18: warning: comparison is always false due to limited range of data type [-Wtype-limits]
  234 |         if (temp < 0) temp = 0;
      |             ~~~~~^~~
```

So I looked at the code in question:

``` c++
224 //  Note that our DAB environment maps the softbits to -127 .. 127
225 //  we have to map that onto 0 .. 255
226 
227 void Viterbi::deconvolve(softbit_t *input, uint8_t *output)
228 {
229     uint32_t    i;
230 
231     init_viterbi (&vp, 0);
232     for (i = 0; i < (uint16_t)(frameBits + (K - 1)) * RATE; i ++) {
233         COMPUTETYPE temp = (COMPUTETYPE)input[i] + 127;
234         if (temp < 0) temp = 0;
235         if (temp > 255) temp = 255;
236         symbols[i] = temp;
237     }
238 
239     update_viterbi_blk_GENERIC (&vp, symbols, frameBits + (K - 1));
240 
241     chainback_viterbi (&vp, data, frameBits, 0);
242 
243     for (i = 0; i < (uint16_t)frameBits; i ++)
244         output[i] = getbit (data[i >> 3], i & 07);
245 }
```

softbit_t is int8, COMPUTETYPE is uint16_t
So line 234 should IMVHO be `COMPUTETYPE temp = (COMPUTETYPE)(input[i]+127);`
The tests in 234 and 235 can then be safely removed, because they can never be true.
A comment about the size and signedness of the used data types might be appropriate to make the casual reader aware of this.

Note that I have absolutely no idea of the mathematics involved, I'm just looking at the issue as a C programmer with a little C++ knowledge ;-)

mpbraendli, 2020-01-04:
This stuff could probably be reworked a bit indeed... I did some experiments using different types in the viterbi decoder to find out if there was a performance difference, and introduced the `softbit_t` type alias. (commit 4496a67f)

At that point I did replace `int16_t` by `COMPUTETYPE = uint16_t`. Before, `input` was `int16_t`, and I reduced it to `softbit_t = int8_t`. The `+ 127` didn't overflow the types before.

`COMPUTETYPE` needs to stay 16-bit wide for the calculation to work, but `temp` not necessarily.

If you do `(COMPUTETYPE)(input[i]+127)`, what will be the type the addition uses? It's not clear because of the implicit casting rules. Better be explicit, and most important, avoid signed integer overflow. I'd favour casting `input[i]` to `int32_t`, clamping the value to 0 <= temp < 256 and then storing into `symbols[i]`.

What do you think of that approach?

seife, 2020-01-04:
I just did not think of or even notice the possible int8_t overflow ;-)
Maybe the easiest solution is to make `temp` simply `int32_t`, like

    int32_t temp = (int32_t)input[i] + 127;
    // input is int8_t, so temp now cannot be < 0 or > 255
    symbols[i] = (COMPUTETYPE)temp;

probably best including the comment ;-)
I have no idea about the performance impact, but I'd guess that at modern CPUs, it is quite possible that int32 performs better than int16_t because of alignment issues.

mpbraendli, 2020-01-04:
Sounds good. I can quickly do this now so that we avoid the overhead of a pull request, or do you want to do that?

seife, 2020-01-04:
just fix it if you get to it, no need for a PR. I'm not proud of that finding ;-)

mpbraendli, 2020-01-04:
Done!

closed 2020-01-04 by mpbraendli