AlbrechtL/welle.io #497: Segfault in OFDMProcessor::start

mpbraendli, 2020-01-04:
Found by @tenzap

Another bug in addition to those 2 above  (https://github.com/AlbrechtL/welle.io/pull/428#issuecomment-547517594) : 
* I also noticed when using SoapySDR with "uninitialised" (test it with an invalid device parameter) then there is a crash when enabling/disabling coarse correction, in the expert setting.

```
Thread 1 (Thread 0x7f632b6eec40 (LWP 31420)):
[KCrash Handler]
#6  0x00007f632faa67bb in raise () from /lib/x86_64-linux-gnu/libc.so.6
#7  0x00007f632fa91535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#8  0x00007f632fe5b983 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007f632fe618c6 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007f632fe61901 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#11 0x000055a1d7456e37 in std::thread::operator= (__t=..., this=0x55a1d97b88a0) at /usr/include/c++/8/thread:153
#12 OFDMProcessor::start (this=0x55a1d97b88a0) at /build/welle.io-next/src/backend/ofdm-processor.cpp:127
#13 0x000055a1d745e357 in RadioReceiver::setReceiverOptions (this=0x55a1d97b8120, rro=...) at /build/welle.io-next/src/backend/radio-receiver.cpp:110
#14 0x000055a1d74300a8 in CRadioController::disableCoarseCorrector (this=<optimized out>, disable=<optimized out>) at /build/welle.io-next/src/welle-gui/radio_controller.cpp:426
#15 0x000055a1d747602b in CRadioController::qt_static_metacall (_o=0x7ffe80f59140, _c=<optimized out>, _id=<optimized out>, _a=<optimized out>) at /build/welle.io-next/build/welle-io_autogen/MOX6SOXE6H/moc_radio_controller.cpp:508
#16 0x000055a1d747759b in CRadioController::qt_metacall (this=0x7ffe80f59140, _c=QMetaObject::InvokeMetaMethod, _id=63, _a=0x7ffe80f560d0) at /build/welle.io-next/build/welle-io_autogen/MOX6SOXE6H/moc_radio_controller.cpp:1087
#17 0x00007f6330ddcb8d in ?? () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#18 0x00007f6330d44421 in ?? () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#19 0x00007f6330d45855 in ?? () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#20 0x00007f6330d465a2 in QV4::QObjectMethod::callInternal(QV4::Value const*, QV4::Value const*, int) const () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#21 0x00007f6330d60a69 in QV4::Runtime::method_callProperty(QV4::ExecutionEngine*, QV4::Value*, int, QV4::Value*, int) () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#22 0x00007f6330d541f2 in ?? () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#23 0x00007f6330e0a4ff in QQmlJavaScriptExpression::evaluate(QV4::CallData*, bool*) () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#24 0x00007f6330d9e527 in QQmlBoundSignalExpression::evaluate(void**) () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#25 0x00007f6330d9f6bb in ?? () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#26 0x00007f6330de75f3 in QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#27 0x00007f6330d81a8a in QQmlData::signalEmitted(QAbstractDeclarativeData*, QObject*, int, void**) () from /lib/x86_64-linux-gnu/libQt5Qml.so.5
#28 0x00007f63301e619f in QMetaObject::activate(QObject*, int, int, void**) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x00007f632f736788 in QQuickAbstractButton::setChecked(bool) () from /lib/x86_64-linux-gnu/libQt5QuickTemplates2.so.5
#30 0x00007f632f736dbf in QQuickAbstractButtonPrivate::toggle(bool) () from /lib/x86_64-linux-gnu/libQt5QuickTemplates2.so.5
#31 0x00007f632f742c75 in QQuickCheckBox::nextCheckState() () from /lib/x86_64-linux-gnu/libQt5QuickTemplates2.so.5
#32 0x00007f632f738caa in QQuickAbstractButtonPrivate::handleRelease(QPointF const&) () from /lib/x86_64-linux-gnu/libQt5QuickTemplates2.so.5
#33 0x00007f632f74fcd5 in QQuickControl::mouseReleaseEvent(QMouseEvent*) () from /lib/x86_64-linux-gnu/libQt5QuickTemplates2.so.5
#34 0x00007f6331170eb8 in QQuickItem::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Quick.so.5
#35 0x00007f633155b4c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#36 0x00007f6331562970 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#37 0x00007f63301bd4f9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#38 0x00007f633118925b in QQuickWindowPrivate::deliverMouseEvent(QQuickPointerMouseEvent*) () from /lib/x86_64-linux-gnu/libQt5Quick.so.5
#39 0x00007f6331189d3b in QQuickWindowPrivate::deliverPointerEvent(QQuickPointerEvent*) () from /lib/x86_64-linux-gnu/libQt5Quick.so.5
#40 0x00007f633118a66f in QQuickWindowPrivate::handleMouseEvent(QMouseEvent*) () from /lib/x86_64-linux-gnu/libQt5Quick.so.5
#41 0x00007f6330575205 in QWindow::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#42 0x00007f633118c2db in QQuickWindow::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Quick.so.5
#43 0x00007f633155b4c1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#44 0x00007f6331562970 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#45 0x00007f63301bd4f9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#46 0x00007f6330568ff3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#47 0x00007f633056ae75 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#48 0x00007f633054506b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#49 0x00007f632b32140b in ?? () from /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#50 0x00007f63301bc1cb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#51 0x00007f63301c41a2 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#52 0x000055a1d741b71d in main (argc=<optimized out>, argv=<optimized out>) at /build/welle.io-next/src/welle-gui/main.cpp:171
[Inferior 1 (process 31420) detached]
```

_Originally posted by @tenzap in https://github.com/AlbrechtL/welle.io/pull/428#issuecomment-554307985_

mpbraendli, 2020-01-04:
I hope commit 544a1b3 actually fixes this.

tenzap, 2020-01-08:
I just tested it with [nightly 7953a9](https://bintray.com/albrechtl/welle.io/download_file?file_path=20200106_7953a9f_Linux_welle-io-x86_64.AppImage) which is posterior to commit 544a1b3 and there is no more crash! :)

Thank you!

closed 2020-01-15 by AlbrechtL