AlbrechtL/welle.io #499: Suggestion: With "dark mode" font of the channel list difficult to read

Rootinator, 2020-01-08:
Hi there,

unfortunately the channel list (left) is very difficult to read in dark mode.
It would be good if the background was designed as the right side or the color of the font changed.

AlbrechtL, 2020-01-15:
@tenzap Can you have a look into it?

tenzap, 2020-01-15:
Which color would you suggest ? Ideally the color should be automatically set, but on this widget, the back/foreground color for text doesn't seem automatic, so that would require a special definition of font color in that case.

Rootinator, 2020-01-25:
Possibility 1:
Leave the background as it is and change the font color from black to white or light gray

Possibility 2:
The same color as in the station name field

Option 3:
Something so that the stations (left) are legible :P

tenzap, 2020-01-26:
Would this be ok ?

* Material Dark
[image]

* Universal Dark
[image]

BTW: I also changed the font color for the switch labels & the combobox values

Rootinator, 2020-01-26:
That looks great! :)

-----

(Because of the SoapySDR error in the lower part of the screenshots: I can still use the SoapySDR in a next version?)

tenzap, 2020-01-26:
> When can I compile a new version with the new font color?

As soon as you apply the [patch of the #503 PR](https://patch-diff.githubusercontent.com/raw/AlbrechtL/welle.io/pull/503.patch ) locally

> (Because of the SoapySDR error in the lower part of the screenshots: I can still use the SoapySDR in a next version?)

No worry, my device wasn't connected, hence the message ;)

Rootinator, 2020-01-26:
What exactly do I have to do to apply the patch? 

Do I have to compile welle.io 2.1 again (without changes) or how do I apply it? :)

tenzap, 2020-01-27:
apply the patch either on the source of 2.1 release (it's unsure whether there will be errors while patching, but I don't think there will be any) or on the source of the next branch.
then, compile it

Rootinator, 2020-01-27:
Sorry, but I do not know which file from welle.io (source code) I have to insert it into.  Can you explain that to me step by step?

tenzap, 2020-01-27:
From within the root of the source folder of welle.io (`welle.io-2.1` for example) run this command
```
patch -p1 < "the_patch_file_you_downloaded.patch"
```
Then compile as usual

Rootinator, 2020-02-01:
Thank you for the patch!
Now I can read everything well.

AlbrechtL, 2020-02-14:
Thanks @tenzap for fixing!

closed 2020-02-14 by AlbrechtL