AlbrechtL/welle.io #501: Window for live Traffic

DG1PW, 2020-01-15:
DAB+ Radio stations provides data for live Traffic or send signal when traffic reports starts.
But there is no window where this information is shown.
Is it possible that the live traffic date can be decode?
Will be very helpful in gives further information which is provided.

andimik, 2020-01-15:
I guess you mean so-called "Service Announcement". This is not only related to traffic information (like via RDS).

DG1PW, 2020-01-15:
Yes, the radio stations provide additional text information. Is there a possibility to encode and display this Announcement?

andimik, 2020-01-15:
Could you please tell us what you mean with "additional text information"?

If you mean TPEG have a look at Qt-DAB, where you get it via localhost in your browser.

AlbrechtL, 2020-01-15:
@andimik can you post a screen shot how it looks like with Qt-DAB?

andimik, 2020-01-15:
You will not be able to read it. This is not text as you might expect.

> Ó,×™·–ÇIã‡âNÒþb&‡´ûÐîrÐW>~5ï…$@óòM�Ô[Ï _¬‚¤Ç~s¿Õ“z5¸æïÈ.}*ÄOfê®³Ú°++Å	‹Tlç›¹ø	¾šð&¹¥iuïÒ@Pôh@~ïÃÛ!ÜÕ=ÞÉ]®¸2!9vrÓèŒ#ÔNp½²¢ê6«q(%ðÐâ!öÞ:ŽKL-Sh
> ÖP-_!Ç8ýclÓÂ$±ñþ”<©~
> L3÷Ïl°–¸?ÄØ
> Ó-ë0áÆºßeúõ	E×¦E¿ÛÐlt?)àæëbÝDiÎ9}šÕø½ûFAúU^äup1sJKÌ½Ì=Ó­T!þUA—ÎlË¸ñ–ÓPPF:c…“´'ör„lô=­U²„fGDÌ!vãˆø9ö4¿zØ’öèwZžwbŽ£ý&y}”‡q(fìë#`íƒ
> )NÍ¹U,7MßŠö*
> çç ›ÅàËõêªQLéÕÛN˜›l}Á~3ª²J÷)Om'õŽ¸äÓî&n}ˆý:E¼qV<ˆO¢’_£6YæÖæ^+¨.õßß¨[œ	åÏ«GÓ”²½FÙÖñ]B#Ã®<dpÏ©%’Åà­àÎân$_?DR¡&ÖOÿæý¿Õ¡ªO€Ð
> Œci>›æÈt[^ÇKJõõ…›r	²èãE*“Ö…ê|ª¤Ó=U”wÞ0kƒöµV.€AvÆ MÛ66B‡#Ë¸IÖåÈå0Ã¥¯núÜ<Æ?á_”æcKÆ0Ú’ã	²ôhÛm&Éæ&ò’šFº(?èž6ÆÀeÑºë¦é¬î†…î”Ug­ÝÅøÊ¬fì$|”æ†¡Ç@ËÁ «ÿFÌXN‚³Ž¸;9£Ä½ø$§ö¬Ã˜QHˆzLª]¶•óh¡Y#Å½j¨ð8"Îñ´yZ°á�_v¢ºp1KsM8£s«¾\þ2®¼÷ÒÖ?h¥c	¥ëÜ(Ål.Bá’ohqK¹kÄø®'L%>Ïæ9’ó|UÓÊWÆ<a(§õØ³™‹»g#,ìU7Cm(Ù\*=<Šr·ÿ2Bñ¶¢¼•˜ÓÜO1¦=@EÕÊm³ÜcSÅIDäÕ;ÿôÌ|ÿ�ÿ����

DG1PW, 2020-01-16:
Yes @andimik  I mean TPEG, but this is not user frindly to do this procedure in QT-DAB.
Here in well.io the TPEG Channel is not shown in the bouquet - why?  And will this not a usefull enhancement to have the TPEG and EPG information?

mpbraendli, 2020-01-16:
welle.io does not yet have the decoder for Packet Data services. I am currently implementing parts that are useful for decoding IP Datagram Tunneling, which are also carried in a Packet Data service, and this could be useful in the future for decoding TPEG.

At the moment, welle.io doesn't show the TPEG service in the list, because it wouldn't bring any benefit as it cannot be decoded.

In addition to this, in order to display the TPEG information in a window, a TPEG decoder needs to be implemented.

EPG (and the related Service Information) is again a different story, which also needs an appropriate decoder.

You also mention 

> send signal when traffic reports starts

which are called Announcements in DAB terminology. welle.io doesn't decode those either.

All these features would be nice to have though.

DG1PW, 2020-01-16:
I asked in the meanwhile a good known technical manager of a broadcast station.
On some channels the TPEG channel is used to provide traffic announcements but it is also possible to provide this information in the FIC / FIDC part of the radio station. So each station can provide there own signal. On TPEG it is the sum of all.

I also found on the www.avt.nbg.de  a unit which have also a FIC-XTractor  (https://www.avt-nbg.de/en/products/dab-xplorer) 

Further Information also here (https://digiradio.ch/linear-digitalradio-begriffe-glossar/)

FIC

Der FIC (Fast Information Channel) wurde hauptsächlich als Systemsteuerkanal konzipiert und beinhaltet alle Informationen über den Inhalt sowie die Struktur des Ensembles. Über den FIC werden Sendernamen, die Zugehörigkeit der PADs zu passenden Radioprogrammen, PTYs und eine Vielzahl weiterer Zusatzinformationen koordiniert. 

FIDC

Der FIDC (Fast Information Data Channel) ist ein Bestandteil des FIC. Er dient für diverse Arten von Anwendungen, welche nicht direkt mit dem Inhalt oder der Struktur des Ensembles im Zusammenhang stehen. Über den FIDC werden beispielsweise Verkehrsinformationen des TMC übertragen.

mpbraendli, 2020-01-16:
As a complement to what you wrote (which is correct): FIDC was removed from the v2 DAB specification (clause 8.2 of ETSI EN 300 401 is now gone)

AlbrechtL, 2020-01-18:
I did a quick Google research. I couldn't find any public available specification about TPEG and EPG. Also there are no open source projects which are implementing such decoders.

Is my assumption correct or did I miss something?

andimik, 2020-02-02:
Qt-DAB exports EPG in so-called `*.ehb` files which is xml. But it does only work on some ensembles (like the German ones), not so in Croatia and Slovenia.

Qt-DAB will also save thumbnails of all stations in the EPG, if they are transmitted in this subchannel.

andimik, 2020-02-04:
Additional remark: The good old Noxon Software from Fraunhofer under Windows (Version 5.01) has a hidden folder `.mmp` in the user's directory where it stores, slideshows, ensemble configuration and even SPI cache files. 

So I found out that in Slovenia and Croatia the EPG is not really existing (let's say they transmit only headers, no contents).

AlbrechtL, 2024-10-09:
I'm closing this issue because the DAB data services are not in the focus right now.

If somebody provides a PR, that would be fantastic!

closed 2024-10-09 by AlbrechtL