AlbrechtL/welle.io #502: 64k HE-AACv2 streams not decoding in welle-cli

whiskerp, 2020-01-24:
Hi
welle-cli doesn't decode a 64k HE-AACv2 stream - it just shows a continuously growing error count on the web page. I changed the stream to 64k HE-AAC in the mux and it decodes fine.
Peter

andimik, 2020-01-24:
I will test it with Capital on 12C DAB Italia (is on satellite 9 East)

http://www.digitalbitrate.com/dtv.php?mux=12C&pid=21017&live=158&sec=0&lang=en
https://www.fmlist.org/sendertabelle/dab-ww_more.php?inputLand=I&inputEn=5009

Or you test it with an older file (but seems Capital was 64 kBit v2 as well) https://github.com/piratfm/eti-tools/files/1822870/12517_dab.ts.zip and use https://github.com/newspaperman/tsniv2ni and ODR-dabmod, if you are an insider.

whiskerp, 2020-01-25:
I have tested it on the UK multiplexes (32k v2) and it does not work on them either. I have used rtlsdr. I also run odr-dab with an easydabv2 so can play with different encoding and bitrates - I see the same problem. welle.io is fine, it is just welle-cli.

andimik, 2020-01-27:
Well, I have just tested it with some DAB ensembles (file and live). I just noticed that there are lots of "superframefilter" errors. Don't know if it comes from welle.io or ODR-DabMod

The error when in the webpage lots of special characters appear can be the case when you started welle-cli in the wrong directory.

In my case it is in `welle.io/src/welle-cli`

Try to change the directory.

andimik, 2020-01-27:
But in general, I cannot confirm your issue. HE-AACv2 stations can be heard like HE-AACv1.

whiskerp, 2020-01-29:
I have compiled on Debian testing (Bullseye) against the latest libraries and have tried master and next versions of welle-cli as well as both FT libraries but in general it will not decode DAB+. The third error digit just climbs rapidly. Using the KISS library sometimes decodes just the first DAB+ stream. I have tried multiple sources including the SDL National broadcast, so it is not an ODR-DAB issue.

andimik, 2020-01-30:
I was just testing again using an error-free ETI and ODR-DabMod.

I also get frame errors and RS errors, but they are very rare. Can listen to any v2 station.

I'm using Ubuntu 18.04 bionic.

Could you share the error messages on the console and post a screenshot?

andimik, 2020-02-02:
Could you please try it out at http://freedabni.tk/ ? I can listen to all stations, LC, v1 and v2.

andimik, 2020-02-02:
Which command do you use to start welle-cli? Try to avoid parameter `D` (for example `Dw` becomes `w`) and replace it by a new parameter `-C 1`.

mpbraendli, 2020-02-04:
I'm sorry, I cannot reproduce. Maybe it is indeed a CPU usage question.

andimik, 2020-02-04:
Well, I just encountered, that the 3 error numbers in the column "Errors" remain 0 when tuning to a DAB only mux. You could try with the above mentioned link and change to 12B. On 5B (DAB+ only) they are growing.

whiskerp, 2020-02-04:
Hi
Ok, I rebuilt this on my raspberry pi 4B (4GB) and it works OK. Strange. Incidentally the PI can cope with encoding 10 channels and running dab-mux as well as welle-cli with -D !!! It pushes it from 25% to 75% busy but I am well impressed!
:-)

andimik, 2020-04-09:
@mpbraendli

I am still not sure what causes these errors. It has nothing to do with reception as I can reproduce it with ODR-DabMod as well.

I fear that either the audio or the slides cause it. But more the audio. 

I tried with an eti from BBC (mp2 only). No errors.
Any other Mux with AAC has errors.

And I tried self created eti files, as long as they are DAB: no errors, but if you change the odr-audioenc to AAC you will see growing errors.

Will post some screenshots.

basicmaster, 2020-05-01:
Possibly related to Opendigitalradio/dablin#61.

basicmaster, 2020-05-01:
@whiskerp 

Could you please post the output of `dpkg -l | grep faad` on the affected system? It seems that recent version of libfaad2 have issues with PS channels; you will only hear audio for the fraction of a second.

AlbrechtL, 2020-07-12:
Any news here?

glokhoff, 2022-03-05:
A station on our local mux wants to transmit with 48 kbps, so I used settings in ODR-Audioenc for 48 kbps with various combinations of sample frequency and PS on/off. This is what I observe in the web page of welle-cli:

FS      PS     result
48     on     no audio, growing error number
48     off     audio OK, no growing error number
32     on     no audio, growing error number
32     off     audio OK, no growing error number

The package info;

libfaad2 2.9.1-1
welle.io  2.1-2build1

Running on a i5 linux mint 20 machine Cinnamon 4.6.7 with just limited load (typically less than 50%)
Hope this helps...

basicmaster, 2022-03-05:
The actual reason is [a bug in specific libfaad2 versions i.e. 2.9.0 and 2.9.1](https://github.com/Opendigitalradio/dablin#ubuntu-2004-faad2-and-he-aac-v2-services). If the package source doesn't provide a fixed version, you will have to compile that library yourself to be able to listen to services with PS.

basicmaster, 2022-03-05:
See also https://github.com/Opendigitalradio/dablin/issues/61

glokhoff, 2022-03-05:
OK, thanks, at least it is clear now what is the reason !

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL