AlbrechtL/welle.io #505: 20200212_e36c57e build not starting on Ubuntu 16.04 LTS

erikdsgh, 2020-02-13:
Just to give feedback:

On Ubuntu 18.04 this latest nightly release works withour problem.

On Ubuntu 16.04 is does not start. Is there something that must be added to Ubuntu 16.04 to allow it to run? 
What is the purpose of the added  20200212_e36c57e_Linux_welle-io-cli-x86_64.AppImage package?

Thanks in advance for any answer.

Erik.

AlbrechtL, 2020-02-13:
Please post the command line output on Ubuntu 16.04.

The usage of welle-cli can be found here: https://github.com/AlbrechtL/welle.io/tree/next#welle-cli

erikdsgh, 2020-02-13:
Please find the command output in the attachment.
The error messages point to GLIBC version needed.
Update under Ubuntu 16.04 seems problematic. In shall stay with the latest version that worked on Ubuntu 16.04 namely 20200127_6a6a459_Linux_welle-io-x86_64.AppImage.
I can use the most recent nightly build on another PC running Ubuntu 18.04.
[Terminal output welle_io.txt](https://github.com/AlbrechtL/welle.io/files/4197551/Terminal.output.welle_io.txt)

AlbrechtL, 2020-02-13:
You are getting the following messages
```
/tmp/.mount_202002upCJ2s/AppRun.wrapped: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.27' not found (required by /tmp/.mount_202002upCJ2s/usr/bin/../lib/libSoapySDR.so.0.7)
/tmp/.mount_202002upCJ2s/AppRun.wrapped: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by /tmp/.mount_202002upCJ2s/usr/bin/../lib/libQt5Gui.so.5)
/tmp/.mount_202002upCJ2s/AppRun.wrapped: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.25' not found (required by /tmp/.mount_202002upCJ2s/usr/bin/../lib/libQt5Core.so.5)
/tmp/.mount_202002upCJ2s/AppRun.wrapped: /lib/x86_64-linux-gnu/libz.so.1: version `ZLIB_1.2.9' not found (required by /tmp/.mount_202002upCJ2s/usr/bin/../lib/libpng16.so.16)
/tmp/.mount_202002upCJ2s/AppRun.wrapped: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.27' not found (required by /tmp/.mount_202002upCJ2s/usr/bin/../lib/libpulsecommon-11.1.so)
/tmp/.mount_202002upCJ2s/AppRun.wrapped: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.25' not found (required by /tmp/.mount_202002upCJ2s/usr/bin/../lib/libsystemd.so.0)
/tmp/.mount_202002upCJ2s/AppRun.wrapped: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.27' not found (required by /tmp/.mount_202002upCJ2s/usr/bin/../lib/libsystemd.so.0)
```
That's the big disadvantage of building AppImages. The supported Linux versions depends on the Linux distribution where it is built (in this case the libc). I switched to Ubuntu 18.04 because it becomes more and more complicated to build welle.io on Ubuntu 16.04 with a modern GCC and current Qt version.

**I do the maintaining in my free time and I don't want to waste time to support 4 years old Linux distributions.**

closed 2020-02-13 by AlbrechtL

erikdsgh, 2020-02-17:
> **I do the maintaining in my free time and I don't want to waste time to support 4 years old Linux distributions.**

I fully understand that. I was not asking for anything, just giving feedback.
In the mean time i continue to use the latest build on Ubuntu 18.04 LTS and as stated this works nicely.
Thanks to your reply i have learned that AppImages  have some drawbacks. 
Good to know.

Best regards.