AlbrechtL/welle.io #506: Release version 2.2 ?

tenzap, 2020-02-16:
Hello,
May I ask for a 2.2 release ?
Some crashes & bugs have been fixed in next since 2.1
Regards

AlbrechtL, 2020-02-29:
Good question. There are no technical issues to not do a 2.2 release. It is just a matter of doing.

In the mean time everyone can use the nightly versions: https://bintray.com/albrechtl/welle.io/welle.io_nightly#files

alexmyczko, 2020-03-02:
a release helps packagers to package a new version, i guess it's too late for Ubuntu 20.04 LTS now.

AlbrechtL, 2020-03-03:
Stupid question: Do we have an Ubuntu package?

alexmyczko, 2020-03-03:
Sure Devuan, Ubuntu, Raspbian just copy the Debian source package

AlbrechtL, 2020-03-03:
Do we have a Debian package?

alexmyczko, 2020-03-03:
Sure, including screenshot:
https://packages.debian.org/sid/welle.io

and here for a general overview for linux distributions:
https://repology.org/project/welle.io/versions

AlbrechtL, 2020-03-03:
Oh great! Thanks for creating this package!!!!

Do you know when Ubuntu copies the sources? I suppose somewhere are the dates.

alexmyczko, 2020-03-03:
https://wiki.ubuntu.com/FocalFossa/ReleaseSchedule

AlbrechtL, 2020-03-03:
OK it was on 27th, February. We missed it.

But thanks for pointing out that DEBs are existing. We should update our website for the Debian repository!

AlbrechtL, 2020-03-03:
**CALL TO ALL**

Please test the latest next version and report any issues. If it is working fine report it in this issue, please.

Source: https://github.com/AlbrechtL/welle.io/tree/next
Nightly builds: https://bintray.com/albrechtl/welle.io/welle.io_nightly#files

andimik, 2020-03-03:
I cannot load a (prerecorded) raw file with `20200302_9091de4_Linux_welle-io-x86_64.AppImage`, so I cannot proceed.

The button "Open RAW file" does not work.

[welle.log](https://github.com/AlbrechtL/welle.io/files/4283880/welle.log)

andimik, 2020-03-04:
Some comments on the latest build

- when changing the language the window resizes
- After switching to another language the PTy is not updated to the new language (e. g. Reiseinformation keeps it in English)
- same with the window title label "Einstellungen" (the rest of the settings menu gets the new language)
- if one starts the station scan (Suchlauf starten) and tunes manually to a channel the number of "Found channels" (Gefundene Sender) in the Service Overview window is wrong (= duplicates are not sorted out)
- in my opinion the refresh rate for the y axis of the spectrum is too fast
- some jpgs in the Slideshow window are not displayed (only the grey asterisk), I've to investigate why, but I fear it has to do with my Ubuntu installation as they are not displayed in Qt-DAB and all my picture tools (Gimp etc.) at all

AlbrechtL, 2020-03-12:
@andimik Thanks for testing. Please open for each of your points a new issue.

tenzap, 2020-03-24:
> After switching to another language the PTy is not updated to the new language (e. g. Reiseinformation keeps it in English)

Could you apply this patch to next and tell me if that fixes the issue ?
[tr_qobject.patch.txt](https://github.com/AlbrechtL/welle.io/files/4373937/tr_qobject.patch.txt)

tenzap, 2020-03-24:
I have prepared a branch with commits fixing some of the translation issues
Commit https://github.com/tenzap/welle.io/commit/59c6b84795dbaff392b2a9e254e9030dfa47a15a
Commit https://github.com/tenzap/welle.io/commit/8556956ad736f169ba646f1419d11276b3144048

@andimik (or anyone else, maybe @mpbraendli  ?) I'm just waiting for someone to confirm the patch `tr_qml.patch.txt` fixes the PTY translation to create a PR.

Thank you

tenzap, 2020-03-25:
For me, latest next (6cf58db) looks good here on debian provided you merge PR #517 which fixes the first 3 issues [reported by andimik](https://github.com/AlbrechtL/welle.io/issues/506#issuecomment-594791084)

andimik, 2020-03-27:
Don't let me open an issue for each bug ... Let's use the task list.

I've tested patch https://github.com/AlbrechtL/welle.io/files/4374539/tr_qml.patch.txt 

- [ ] when changing the language the window resizes

still the case
but if you have the full screen mode on, then the window is cropped (see left column with the station list)

before 

[image]

after

[image]

- [ ] After switching to another language the PTy is not updated to the new language (e. g. Reiseinformation keeps it in English)

Don't remember if it was the case before, but now it loses the lock, when I click on the same service like before, the PTy language is updated

- [x] same with the window title label "Einstellungen" (the rest of the settings menu gets the new language)

Yes, confirmed, this patch fixes it.

andimik, 2020-03-27:
- [x] The button "Open RAW file" does not work.

No idea why, but it works again (see screenshots above)

tenzap, 2020-03-27:
The patch I mentioned was mainly for the translation of the PTY. It didn't include all fixes. If you want to test the fixes and changes I made recently, please have a look at these PR:
#517 #520 #521 (#522)

andimik, 2020-03-27:
- [x] After switching to another language the PTy is not updated to the new language (e. g. Reiseinformation keeps it in English) https://github.com/AlbrechtL/welle.io/pull/517
- [x] but now it loses the lock 
- [x] when changing the language the window resizes

These bugs are fixed in your fork. Good job! PRs should be merged.

Regarding https://github.com/AlbrechtL/welle.io/pull/521 I don't understand why this should be solved. When I change the format (e.g. from u8 to s8), the spectrum window is frozen, I have to click on a service (radio station) or start the station scan, only then it reloads (and plays, if the format is the right one).

- [ ] gui: (re)load RAW file as soon as selected or format is changed

tenzap, 2020-03-27:
Initillay I didn't understand how to make this "raw file" work, because nothing was happening after selecting a file. That was very confusing to me. Later on, on noticed that the file was only loaded when closing the "settings" dialog which I found not intuitive since selecting another device immediately triggers it loading. So I did #521 to have the file loaded as soon as selected, or as soon as the user selects another format.

Concerning your problem, are your backend settings in the expert settings adapted to your raw file ? I also faced some playback issues, but couldn't do better. It was kind of random...

Do you have these problems also with v2.1 or v2.0 ?

tenzap, 2020-03-28:
Please see #524 . I guess this will fix what you describe with RAW files (if you have autoplay enabled & already play one station of that file, modifying the format for example will trigger a reload of the file & play that same station again)
My fork includes the fix if you want to give it a try

andimik, 2020-03-28:
No, I cannot confirm, have tried 5a91800. After selecting a raw file a user still has to start a scan. There is no auto-loading like you describe. 

It would even be better to clear the station list before scanning, because otherwise you have the services from the previous file in the station list and **cannot play them** (as the previous file is not loaded).

Another nice-to-have fix: When playing a raw file, it makes sense to hide the "manual channel". Because if you change to another channel, nothing happens of course and you have to re-select one service from the list.

[image]

BTW: 

Seems the text in the task bar does not like HTML tags at all, see my test

[image]

tenzap, 2020-03-28:
> I have tried [5a91800](https://github.com/AlbrechtL/welle.io/commit/5a91800261a5bc50eeadc5f6d7835f9ef58d44f2). After selecting a raw file a user still has to start a scan. There is no auto-loading like you describe. It only works when you are working with the same files all the time For testing this is ok. Let's say, it's a kind of developer feature ...

Yes indeed, this is in someway what it is. Actually it doesn't need to be the same file, I can be another one, provided it was recorded on the same DAB "ensemble" (scanning is then not needed, and it will restart playing on the same channel).

Concerning the other points, I have no interest in working on them. Could you please create new Issues for each them ? Then I will post some comments. Thanks

AlbrechtL, 2020-03-30:
Thanks for all your comments and PRs!

At the moment I'm moving with my family into a house so my top priority is packing our stuff and organizing it during a difficult time.

Please be patient. In a silent minute I will review all your comments and PRs.

AlbrechtL, 2020-07-11:
I would like to publish the version 2.2.

So I need your help for testing. Is the nightly version `20200609_ed4d5d5_xxxx ` (https://bintray.com/albrechtl/welle.io/welle.io_nightly#files) working?

@tenzap, @andimik, @alexmyczko Please test this version. If there is a major issue which prevents the version 2.2 please report it.

andimik, 2020-07-12:
Well, I've tested both 

20200609_ed4d5d5_Linux_welle-io-x86_64.AppImage
and
20200711_9f4342b_Linux_welle-io-x86_64.AppImage

but will refer to the following to 9f4342b

- [ ]  when changing the language the window resizes (is fixed in tenzap's fork as far as I know)

- [ ] Autodetection of rawfile format does not work properly (sdr 16bit are displayed as 8bit)
[image]

- [ ] Waterfall diagram for 16bit files seems not to work properly (= I just saw it once)
[image]
[image]

- [ ] new severe bug found when 32kHz and 48kHz sampling rate are used in the same mux, see  https://github.com/AlbrechtL/welle.io/issues/573

Some cosmetic changes:

- [ ] When playing a raw file, it makes sense to hide the "manual channel". 

- [ ] the field strength icon has nearly the same colors
[image]
the grey should better be white, inverse or even red

- [ ] If there is no TII, please use a threshold or even fix the y-axis (amplitude)
[image]

- [ ] When I chose another service the slideshow still is shown for the old service (I guess this only works correctly when the new service has SLS as well)

[image]

- [ ] there is no information where MOT Slideshows are stored after "save all images" dialogue.
I only see it in the console
> 2020-07-12T10:58:23.074 Info: SLS: Saving picture "/home/andreas/MOT_0__0_Soundset37537.jpg"

- [ ] in "Material" style all title bars (= also in the settings dialog) have the wrong color (black instead of white)
[image]

AlbrechtL, 2020-07-12:
@andimik Thanks for testing. To me it is very hard to track all your points inside this issue. To be really sure that we don't miss a bug the best is to open an issue for each bug.

andimik, 2020-07-14:
done, opened several issues (some are not a bug, but users might expect other behavior)

alexmyczko, 2020-07-30:
https://en.wikipedia.org/wiki/Release_early,_release_often

alexmyczko, 2020-08-26:
guess you can close this now when it was released several days ago?

AlbrechtL, 2020-08-31:
@alexmyczko Thats' correct!

closed 2020-08-31 by AlbrechtL