AlbrechtL/welle.io #510: no tuned channels visible in the window

rikri, 2020-02-28:
Hi!
I'm not successful with tuning of channels. On the terminal output I see that something has been found, I also see counting of founded channels during scanning...(see bellow) But when it is done I see just window with "No Station". Do I do something wrong?
Thank you.
Richard
[image]
[image]

andimik, 2020-02-28:
Where do you live? You were searching in the L-Band as well which is only used in Brno (CZ) and Vatican City.

rikri, 2020-02-28:
This could be the reason... ;-) I live In Praha. The truth is that I just "pressed the button" and then looked how channels are found and then... nothing. What also confused me were channels that I was seeing in the output in console:

`020-02-28T13:38:02.142 Info: OFDM-processor: reset (0)
2020-02-28T13:38:02.142 Info: OFDM-processor:start
2020-02-28T13:38:02.163 Info: ofdm-processor: Lost coarse sync (coarseCorrector: 0; fineCorrector: 0)
2020-02-28T13:38:02.511 Info: ofdm-processor: Found sync (coarseCorrector: 0; fineCorrector: 0 after 5 frames)
2020-02-28T13:38:02.642 Debug: RadioController: ID of ensemble: 8197
2020-02-28T13:38:02.748 Debug: RadioController: Label of ensemble: "CRo DAB+        "
2020-02-28T13:38:03.205 Debug: RadioController: Found service  2F1B "CRo-RETRO       "
2020-02-28T13:38:03.313 Debug: RadioController: Found service  2608 "CRo-PLZEN       "
2020-02-28T13:38:03.430 Debug: RadioController: Found service  243F "CRo-REGION      "
2020-02-28T13:38:03.547 Debug: RadioController: Found service  2F23 "CRo-WAVE        "
2020-02-28T13:38:03.665 Debug: RadioController: Found service  2F27 "CRo-JAZZ        "
2020-02-28T13:38:03.789 Debug: RadioController: Found service  232E "CRo-DVOJKA      "
2020-02-28T13:38:03.910 Debug: RadioController: Found service  2F25 "CRo-JUNIOR      "
2020-02-28T13:38:04.022 Debug: RadioController: Found service  2F26 "CRo-DDUR        "
2020-02-28T13:38:04.158 Debug: RadioController: Found service  2424 "CRo-PLUS        "
2020-02-28T13:38:04.275 Debug: RadioController: Found service  232D "CRo-VLTAVA      "
2020-02-28T13:38:04.385 Debug: RadioController: Found service  232F "CRo-RADIOZURNAL "
2020-02-28T13:38:04.506 Debug: RadioController: Found service  250D "CRo-C.BUDEJOVICE"
2020-02-28T13:38:04.625 Debug: RadioController: Found service  280C "CRo-PARDUBICE   "
2020-02-28T13:38:04.736 Debug: RadioController: Found service  2706 "CRo-LIBEREC     "
2020-02-28T13:38:04.839 Debug: RadioController: Found service  2F1A "CRo-DAB PRAHA   "
2020-02-28T13:38:11.746 Debug: RadioController: Tune to channel "12D" -> 229.072 MHz
2020-02-28T13:38:11.751 Info: OFDM-processor: input not ok, closing down
2020-02-28T13:38:11.840 Info: End RTLSDR thread
2020-02-28T13:38:11.901 Info: Start RTLSDR thread
Allocating 15 zero-copy buffers`

That seemed to me that program has found something and I just don't see it.
Another fact is that my hardware is bellow minimum requirements - I'm trying it on C2D T7200. Does it make sense on such hardware? Could this be the problem?
Thank you for your time.

AlbrechtL, 2020-02-29:
Thanks for your report! What is a C2D T7200?

rikri, 2020-02-29:
It is an old Core 2 Duo T7200 processor with just 2 cores that I have in my old laptop. Although it has 2 cores it still should have more power than e.g. RPi2. But it maybe lacks some different properties.(?)
I have hoped that I could use the laptop not just for occasional scripting but also for listening of DAB. No, no, let me reformulate the last sentence  - "I still hope that..." ;-)
Thank you.

AlbrechtL, 2020-02-29:
Can you record a RAW file (https://www.welle.io/devices/rawfile)?

With that I can check if welle.io has an issue.

rikri, 2020-02-29:
OK. I'll do it on Monday. Thank you.
Wish you a nice weekend.

andimik, 2020-02-29:
I guess the raw file will play well.

rikri, 2020-03-03:
I hope I have understood well.
I used this command: `dab_raw_record.sh -c 12C -t 10` to create this raw file [https://we.tl/t-P8oVwuWOc7](url)
Thank you.

AlbrechtL, 2020-03-03:
Thanks for the file.

The signal seems to be very weak. Please have a look into `dab_raw_record.sh`. In line 128 you can find the `rtl_sdr` command. Please add the option `-g` for the gain.
https://github.com/AlbrechtL/welle.io/blob/19ce3d91f1084a24a0ea4c81fa80579a23b9a647/download/dab_raw_record.sh#L128 

Please see also the rtl_sdr manpage: https://manpages.ubuntu.com/manpages/trusty/man1/rtl_sdr.1.html

andimik, 2020-03-04:
I saw in your picture that you were changing the drop down menu to "Favorites"

Please change to "All stations"

rikri, 2020-03-05:
> I saw in your picture that you were changing the drop down menu to "Favorites"
> 
> Please change to "All stations"

Uff... That's it. Problem solved. Thank you andimik very much. I'm sorry for wasting time of both of you.
If I may speak up for _notsobright_ users like myself - it would be great to have "All stations" as a default view. 
Thank you once again.

AlbrechtL, 2020-03-06:
Really the favorite view is on by default? If so then it is a bug.

@andimik Can you verify it please?

andimik, 2020-03-27:
@AlbrechtL 

Hallo Albrecht, I've tested it, have deleted `~/.config/welle.io/welle.io.conf` and built from sources. 

**Default is "All stations".** 

You may close this issue.

rikri, 2020-03-27:
Thank you. Maybe I have changed the view by myself during desperate clicking to find some stations and having weak signal first (as AlbrechtL was writing) then when I moved with antenna I haven't seen anything because of changing of view... 
Nevertheless I'm enjoying listening already three weeks so the story has ended with happy end. :-)
Thank you once again and wish a good health to you and your families!
Richard

closed 2020-03-27 by rikri