AlbrechtL/welle.io #514: Please add compatibility with Navit.

Supermagnum, 2020-03-03:
I believe that Navit uses the TRAFF format for traffic information. TraFF is an XML-based format. 
Links:
https://traffxml.gitlab.io/
https://www.navit-project.org/

andimik, 2020-03-03:
Pull Requests are welcome of course.

If you have documents about TPEG, please share them with us.

I don't any free or open source software which can extract TPEG data from the DAB stream.

Supermagnum, 2020-03-04:
https://www.worlddab.org/dab/technical-specifications

AlbrechtL, 2020-03-04:
From what I understood is that traffxml is a format to exchange traffic information. It can be used to transfer this information to other apps like Navit.

But to use that we need a TPEG decoder that we don't have. Do I'm right?

Supermagnum, 2020-03-04:
> From what I understood is that traffxml is a format to exchange traffic information. It can be used to transfer this information to other apps like Navit.
> 
> But to use that we need a TPEG decoder that we don't have. Do I'm right?

Yes, a TPEG decoder is needed. 
Slightly off topic: There are two different TPEG services in Norway, one unencrypted by NRK and one encrypted by v-traffic.

AlbrechtL, 2020-03-06:
If anyone knows a TPEG decoder please let us know it.

Supermagnum, 2020-03-06:
> If anyone knows a TPEG decoder please let us know it.

They are working on it, but they need open unencrypted TPEG raw data:
https://github.com/JvanKatwijk/qt-dab/issues/156

AlbrechtL, 2020-03-07:
Lets see how Jan (@JvanKatwijk) proceeds with his activities.

AlbrechtL, 2024-10-09:
I'm closing this issue because the DAB data services are not in the focus right now.

If somebody provides a PR, that would be fantastic!

closed 2024-10-09 by AlbrechtL