AlbrechtL/welle.io #515: Crash when enabling TTI decoding

tenzap, 2020-03-24:
With latest next 0e498c1 (2020-03-16) there is a crash when enabling TTI decoding in the expert settings

Console output
```
SuperframeFilter: Superframe sync started...
SuperframeFilter: Superframe sync succeeded after 2 frame(s)
AACDecoder: using decoder 'FAAD2'
2020-03-24T08:58:03.829 Debug: RadioController: Audio sample rate 48000 Hz, mode= "HE-AAC, 48 kHz Stereo @ 88 kBit/s"
2020-03-24T08:58:07.225 Info: New Receiver Options: TII: 1 disable coarse corr: 1 freqsync: GetMiddle fft placement: StrongestPeak
terminate called after throwing an instance of 'std::out_of_range'
  what():  PRS length: 0 vs 2048
KCrash: crashing... crashRecursionCounter = 2
KCrash: Application Name = welle-io path = /usr/bin pid = 20996
KCrash: Arguments: /usr/bin/welle-io 
KCrash: Attempting to start /usr/lib/x86_64-linux-gnu/libexec/drkonqi from kdeinit
sock_file=/run/user/1000/kdeinit5__0
2020-03-24T08:58:07.287 Info: RTL_SDR: End rtlsdr_read_async_wrapper() thread
2020-03-24T08:58:07.287 Info: OFDM-processor: input not ok, closing down
```
Kde crash log [full log: welle-io-20200324-085807.kcrash.txt](https://github.com/AlbrechtL/welle.io/files/4373889/welle-io-20200324-085807.kcrash.txt)
```
Thread 15 (Thread 0x7f00ccff9700 (LWP 21020)):
[KCrash Handler]
#6  0x00007f011447e7bb in raise () from /lib/x86_64-linux-gnu/libc.so.6
#7  0x00007f0114469535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#8  0x00007f0114833983 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007f01148398c6 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007f0114839901 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#11 0x00007f0114839b34 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#12 0x000055767bc46e82 in TIIDecoder::run (this=0x55767e02da40) at /usr/include/c++/8/ext/new_allocator.h:86
#13 0x00007f0114862b2f in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#14 0x00007f0116291fa3 in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#15 0x00007f01145404cf in clone () from /lib/x86_64-linux-gnu/libc.so.6
```

mpbraendli, 2020-03-25:
Indeed, it does crash here too. I'm looking into it.

mpbraendli, 2020-03-25:
Should be fixed in 6cf58dbd, thanks for letting us know.

Please open the issue again if necessary.

closed 2020-03-25 by mpbraendli

tenzap, 2020-03-25:
Looks good, thanks !