AlbrechtL/welle.io #516: welle-io: playback stops when enabling coarse correction

tenzap, 2020-03-24:
With latest next 0e498c1:

When "enabling Coarse correction" (with algorithm set to GetMiddle) the backend seems to block/freeze in some way. The effect is the playback stops. And the log remains on
`SuperframeFilter: Superframe sync started...`

To get it playing again, either disable coarse correction, or choose another algorithm.

BTW, it seems, the change of algorithm isn't really taken into account and the backend doesn't seem to react because there is no other output than :
```
2020-03-24T15:01:37.290 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: GetMiddle fft placement: StrongestPeak
2020-03-24T15:01:45.128 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: PatternOfZeros fft placement: StrongestPeak
```

mpbraendli, 2020-03-25:
Hi,

I can't reproduce what you observe: all coarse correction algorithms seem to work here. Maybe it just means that for the receiver you have and the signal you want to receive, the other algorithms work better.

I don't think there is any action to be taken on this topic, so I'll close the issue, but feel free to continue the discussion.

closed 2020-03-25 by mpbraendli

tenzap, 2020-03-25:
At least it doesn't seem to be a regression because there is the same behaviour with 2.1

However, if I enable Coarse correction with a default of GetMiddle, playback stops, then I select another algorithm, playback starts again, and if I select GetMiddle again playback still continues (since it stopped initially I expected it to stop too)

mpbraendli, 2020-03-25:
It's possible that one algorithm is able to keep the reception going, but isn't able to acquire reception after a cut. Maybe that's the case here.