AlbrechtL/welle.io #518: [welle-cli] SLS picture is not updated

tenzap, 2020-03-25:
In welle-cli web interface, the picture of the SLS is not updated when the station sends a new one.

Typically, I listen to a radio that sends a new picture for every new song, but the picture when clicking on SLS doesn't update. It remains the one that was set initially when playing the station.

It seems to be related to the cache in the browser because :
- killing welle-cli and hitting play doesn't update the picture (while leaving the web page displayed in the browser)
- refreshing the page in the browser does the trick.

mpbraendli, 2020-03-25:
Indeed, while the HTTP headers include `Last-Modified`, there is no cache-control header like for the other entries. Can you try with 573fbac and close the issue if it works?

tenzap, 2020-03-25:
The timestamp under the picture gets updated (which was already the case), but the picture still remains the same (tested on firefox & chrome)

mpbraendli, 2020-03-25:
Maybe we still need to do the ugly trick of modifying the img URL. Can you try with the following patch applied:

```patch
diff --git a/src/welle-cli/index.js b/src/welle-cli/index.js
index d34cad9..0dfce69 100644
--- a/src/welle-cli/index.js
+++ b/src/welle-cli/index.js
@@ -220,7 +220,7 @@ var slideimg = document.getElementById("slideimg");
 var slidecaption = document.getElementById("slidecaption");
 
 function showSlide(sid, last_update_time) {
-    slideimg.src = "slide/" + sid;
+    slideimg.src = "slide/" + sid + "?cachebreak=" + last_update_time;
     var last_update = new Date(last_update_time * 1000);
     slidecaption.innerHTML = last_update;
     slide_modal.style.display = "block";
```

(Aside: I really hope one day an experienced web developer takes over and replaces this kludge with something nice instead...)

tenzap, 2020-03-25:
This fixes the issue thanks.

andimik, 2020-03-25:
Confirmed. This fixes it.

That's the new `index.js` 
[index.js.zip](https://github.com/AlbrechtL/welle.io/files/4383389/index.js.zip)

No need to rebuild, just replace it in the build-directory.

mpbraendli, 2020-03-26:
Thanks for confirming it works. It's in 5e59ea2

closed 2020-03-26 by mpbraendli