AlbrechtL/welle.io #519: Support for 2nd PTy (static and dynamic)

andimik, 2020-03-25:
Today I found out that the German Ensemble **SWR BW N** (Südwestrundfunk Baden-Württemberg Nord) has an interesting feature: There are two different PTy for the same service.

[swr_n.eti.zip](https://github.com/AlbrechtL/welle.io/files/4383395/swr_n.eti.zip)

If you play that eti-file (using odr-dabmod) in `welle-cli` the PTy of **SWR4 S** (Stuttgart) seems to "blink" between nothing and Talk (= Varied), all other SWR4 services blink between Talk and News during the world news.

[image]

According to dablin_gtk there are two PTy: one is static, one is dynamic.

[image]

So, in general, welle-cli reads out both PTy fields, but I would suggest to write them in two lines.

For welle-io it seems only the dynamic PTy is displayed (= SWR4 S shows no PTy)

basicmaster, 2020-03-26:
So far I have seen this situation (both dynamic and static programme type for a service) only in the SWR ensembles.

Note that in EN 300 401 v1 this was kind of allowed (i.e. not exactly defined; at least not explicitely forbidden), but in v2 it is now clearly forbidden to have both kind of types for a particular service.

AlbrechtL, 2020-07-12:
Since this behavior is forbidden now I can close this issue.

closed 2020-07-12 by AlbrechtL