AlbrechtL/welle.io #534: No Reception with DVB-T+DAB+FM Stick

johannes-krude, 2020-04-02:
I have bought two different DVB-T+DAB+FM Sticks from two different vendors. Both use a FC0012 Tuner.

When doing a station scan, no stations are found. This does neither work on the ubuntu-20.04 version of welle.io nor on the git next branch.

When trying rtl_fm I am able to hear radio stations but always get a high frequency noise, I am not sure whether I correctly connected rtl_fm to pulseaudio.

Should welle.io work with these sticks? Is there something I need to do to make this work?

andimik, 2020-04-02:
I suppose you live in or Aachen. So there is DAB+ of course in the air.

Seems your laptop or a LED light is producing much noise as FM is effected as well.

Try to use another antenna.

Or post a screenshot of the expert view of Welle.io and the spectrum view. For example for Bundesmux 5C.

johannes-krude, 2020-04-02:
Thanks for looking at my problem.

I live in Aachen. I have [this](https://www.amazon.de/Tragbarer-Mini-Digitaler-TV-Stock-Unterstützung-SDR-Stimmer-Empfänger/dp/B013Q94CT6) stick which is advertised as having an R820T2 tuner, but rtl_test identifier a Fitipower FC0012 tuner. I also have a similar looking stick with a black case.

I know my Laptop is noisy for my USB microphone when plugged into the charger, so I disconnected the charger, but this did neither improve FM or DAB.

I currently only have two instances of the same antenna.

Here is a screenshot of channel 5C, I hope that helps in identifying the cause of the problem.
[image]

tenzap, 2020-04-02:
Have you tried changing the backend settnigs in the "expert settings" ?

johannes-krude, 2020-04-02:
I have now tried changing the backend settings. I tried all 9 possible combinations of Coarse corrector algorithm and FFT Windows placement algorithm. None caused me to have any reception.

andimik, 2020-04-02:
What about the gain?

And what happens when you plug in the cable which normally is used for the TV set?

johannes-krude, 2020-04-03:
The gain was set to automatic, none of the manual gain settings make any difference. Enabling the bias tee also makes no difference.

I didn't quite get the question about the cable TV. All my previous attempts were made with the antenna connected to the stick. I cannot connect any cable TV link to the stick since I do not have cable TV. For comparison I attached the spectrum when nothing is plugged into the antenna port of the stick.
[image]

Maybe I should look for another device. Is there any particular rtl-sdr device you can recommend for DAB+ reception with welle.io?

tenzap, 2020-04-03:
Did you try outside, or at least placing the antenna out of a window ? (I read somewhere that I Germany, DAB+ signal emission was more adapted to outside reception rather than inside)

Maybe try with another antenna ? I made one myself based on this video: https://youtu.be/WP1-IegL2Rs

johannes-krude, 2020-04-03:
I now connected a single 40 cm wire to the antenna connector. This increased the amplitude on 5C to about 70 but did not lead to any reception. There is no change when going onto my balcony.

tenzap, 2020-04-03:
Try the suggestion in the video, if I remember well, it really makes a difference.
Or connect your stick to the TV connector in your wall (connected to the tv/DVB-T antenna on top of your house/building) if there is one...

andimik, 2020-04-03:
Well, I have tried with an old FC0013 (from the same manufacturer, not the same, but similar)

```
$ rtl_test 
Found 1 device(s):
  0:  NOXON, DAB Stick, SN: 0

Using device 0: Terratec NOXON DAB/DAB+ USB dongle (rev 1)
Found Fitipower FC0013 tuner
Supported gain values (23): -9.9 -7.3 -6.5 -6.3 -6.0 -5.8 -5.4 5.8 6.1 6.3 6.5 6.7 6.8 7.0 7.1 17.9 18.1 18.2 18.4 18.6 18.8 19.1 19.7 
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 112 bytes
^CSignal caught, exiting!
```

Indeed, it's a bit tricky to get audio as there is a huge DC spike in the middle of the spectrum, but this is my only DAB signal, 70 km from here from Slovenia with a roof antenna as there is no other DAB signal in Southern Austria yet.

[image]

And this is Qt-DAB
[image]
still a DC spike, but not so huge

In your screenshots there is also the spike and I fear this might cause the issue. Could you try the app image from https://github.com/JvanKatwijk/qt-dab/releases ?

Maybe @mpbraendli could us explain about this behaviour?

AlbrechtL, 2020-05-29:
I guess the differences that you discovered are coming from different gain settings. Please make sure that you are using the same gain settings.

AlbrechtL, 2020-07-11:
Closed because of inactivity.

closed 2020-07-11 by AlbrechtL