AlbrechtL/welle.io #541: CPU Usage remains ~5% after call to CRadioController::stop()

tenzap, 2020-04-09:
PR #532 reveals this:

- When clicking "stop" (which calls CRadioController::stop() ) CPU usage goes down from ~35% to 5% with RTLSDR & Raw file, but with SoapySDR it goes down to  ~0% (I didn't test the other devices). 
- So I guess, something is not being stopped in some cases.
- I noticed that with SoapySDR there is this line in the log `OFDM-processor: input not ok, closing down` which seems to mean it stops the OFDM-processor
- So I tried to add ```radioReceiver->stop();``` to `CRadioController::stop()` https://github.com/AlbrechtL/welle.io/blob/8fc57390bfb1f312094425a329ff52635b405c17/src/welle-gui/radio_controller.cpp#L234 and this actually brings CPU usage to ~0% for RTLSDR & RawFile.

But I don't know whether 
- this is the good thing to do
- other things have to be added (any other check, things to do...), any side effects ?
- where exactly to add this call inside `CRadioController::stop()`

@mpbraendli what is you opinion on this, what do you suggest ?

_Originally posted by @tenzap in https://github.com/AlbrechtL/welle.io/pull/532#issuecomment-606460067_

alexmyczko, 2020-05-11:
I've got a load average of 1.5 (htop shows 25% on a quad core machine, Intel(R) Core(TM) i5-2400S CPU @ 2.50GHz) of 2011.

tenzap, 2021-02-27:
it is fine now since play/stop feature has been merged. closing

closed 2021-02-27 by tenzap