AlbrechtL/welle.io #547: [welle-gui] regression: MOT image is not updated in some cases

tenzap, 2020-04-15:
Between 2.1 & now, at some time there appeared a regression, probably during the MOT SLS commits done in January

_Steps_
Play a station that provides an image having path image://sls//0//0/0000.jpg
Then switch to another station which has the same image path

_Result_
The image that is displayed, will actually be the one of the 1st station.

_Expected result_
Should be the picture of the current playing station

_BTW_
If I play a station whose image has a different name, image is well set.

AlbrechtL, 2020-05-29:
Thanks for reporting!

AlbrechtL, 2024-10-09:
I have issues to reproduce this issue. If a new service is selected, the MOT picture buffer will be cleared. Maybe it is fixed by accident in the last 4 years, or I didn't find it.

For now, I'm closing it.

closed 2024-10-09 by AlbrechtL