AlbrechtL/welle.io #553: [welle-cli] Improvements for JSON (mux.json)

andimik, 2020-04-21:
An Irish DAB+ service provider told me that the traffic from (and with) **welle-cli** has reached a severe load in his network, so I suggest improving it:

Please excuse my idea, but I wonder why the `mux.json` - which is used as input for the webpage - really needs so many fields (lots of them are hidden for the users).

I've analysed the generated network traffic with Firefox' Web developer extensions and found out that every second the complete `mux.json` (in my case here it's 20 kB) is sent over the network. 

Why not split the JSON output into two parts?

a) keep the existing (very detailed) `mux.json` which can be downloaded on request like today
b) create a second JSON with only those fields needed for the webpage. Here, too long descriptions should be avoided and fields should be combined (like we have now with s.mode = Services Mode).

Therefore, the JSON file would be much smaller.

And maybe it can be even more optimized by separating the second, smaller JSON into 

a) nearly static (like Ensemble name, bitrates, labels, protection level etc.) -> updated let's say every 5 seconds or only on change
b) very dynamic (errors, audio level, DLS, SLS) -> updated every second

BTW: The field sub channel ID (subchid) is also missing in the webpage, I just recognized ...

mpbraendli, 2020-04-22:
Hi, it's true that the json contains much more information than the webpage needs, because the webpage was only meant as an illustration of the possibilities. If the web view starts to become a feature in itself, which is apparently the case, it makes sense to rethink the communication.

I'd like to keep the existing mux.json as-is, because I have already written a few scripts that parse it and for which the extensive collection of information makes sense.

But nothing (except maybe the annoyingly long compile time of the relevant C++ files) stands in the way of adding new json URLs with a subset of the information, and adapting the web view to use those.

mpbraendli, 2020-08-11:
I'll close the topic unless someone intends to work on that.

closed 2020-08-11 by mpbraendli