AlbrechtL/welle.io #555: audio level too high, distorted on new install of another mac, under Mojave

Observerist, 2020-05-03:
With plenty of indoor time, I installed welle on another mac, an old, used but fully loaded 2014 mac-mini. That mini runs under Mojave and I also installed GQRX and CubicSDR.

Welle.io has a problem with it's audio output, way too high, totally distorted on the stations that were already known to modulate close to the edge, while the "shy' ones seem to have now the upper hand.
GQRX & CubicSDR, my analog SDRs of choice, don't have any audio bridging problems.

I read that there were plans to introduce a level slider, together with as pause/stop button.
Any news on an upcoming version with this internal audio level control ?

One example I often use to show how a great audio level handling could look like is an internet radio app called Radium. It 'lives" in the top menubar of macOS and has an internal audio routing with individual volume sliders for every playout target that is already "declared" by the system.
It's a "One Source to Many Outputs" setup.
( it's the old version 3.0.10 of Radium, the major update they presented 5 years ago was a no-go )

Here are 2 screenshots of this "reverse output mixer". Just some food for thought. 
Getting ONE slider would already be a grrrreat improvement.

Oooops 
Can't get those small images to attach to this message.

tenzap, 2020-05-05:
> I read that there were plans to introduce a level slider, together with as pause/stop button.

See PR #532 (waiting for merge)

Observerist, 2020-05-05:
#532 ... that's what I read, it was your contribution. 
Really like your simulation. Just hoping for the best and waiting.

AlbrechtL, 2020-07-12:
> Welle.io has a problem with it's audio output, way too high, totally distorted on the stations that were already known to modulate close to the edge, while the "shy' ones seem to have now the upper hand.

I don't understand this behavior. welle.io only deals with digital samples and inside welle.io there is no code which can distort the samples. I assume you have an issue with your analog audio output. Can you double check this? An audio volume slider will only workaround this.

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL