AlbrechtL/welle.io #557: Welle-cli stream entire mux on network

noworthwhile, 2020-05-19:
Dear, 

I'm trying to build a Dab/Dab+ to LAN restreamer setup, similar to Dvblast or Mumudvb for TV. 

I'm having issues figuring out if Welle-cli can tune an entire Mux and stream every station over the network?

I'm trying to build a .m3u with links to each station. 

Is this something possible with Welle-cli?

Thanks.

alexmyczko, 2020-05-19:
welle.cli can do exactly that, see the screenshot here:

https://screenshots.debian.net/package/welle.io

(2nd)

mpbraendli, 2020-05-19:
Hi, if you want to build a m3u playlist you can parse the urls from `/mux.json`, but welle-cli itself doesn't create an m3u file itself.

noworthwhile, 2020-05-19:
Thanks to you both.

Here in Brussels, there are four muxes. I imagine I would need 4 tuners. Should each tuner be connected to one instance of welle-cli? Or can I configure one instance of wells-cli to use all four tuners?

What does the stream look like? Is it HE-AAC?

Thanks.

mpbraendli, 2020-05-23:
Yes, you'd need four instances of welle.io. You'd need a powerful system to decode all programmes of 4 ensembles simultaneously though.

welle-cli re-encodes to mp3, because that's what browsers support.

noworthwhile, 2020-05-23:
> Yes, you'd need four instances of welle.io. You'd need a powerful system to decode all programmes of 4 ensembles simultaneously though.
> 
> welle-cli re-encodes to mp3, because that's what browsers support.

Any way to disable the re-encode? HE-AAC is pretty well supported. It would lower the CPU footprint as well.

andimik, 2020-05-23:
LOL, I just wanted to ask the same question ...

I have no idea if MP4 ADTS will be played like LOAS/LATM.
https://en.wikipedia.org/wiki/HTML5_audio

Or at least to re-encode only after someone pressed PLAY (= on demand, GET)?

mpbraendli, 2020-05-23:
I doubt it's well supported because AAC+ in DAB+ uses a 960 transform size, and not the common 1024. In any case, a proper framing needs to be added around the raw AAC+ AUs, and that framing must be flexible enough to tell about the 960 transform size. I imagine it should somehow be doable in MP4 but I don't know.

When this feature was initially implemented, only mp3 and wav were supported accross all browsers (See HTML5 audio compatibility matrix)

When `-D` is not given, AAC+ decoding and mp3 encoding only gets done when somebody listens the stream. When `-D` or `-C` is given, AAC+ decoding is done, but mp3 encoding only if somebody listens.

AlbrechtL, 2020-07-12:
Can we close this issue?

AlbrechtL, 2024-10-08:
Closed because of inactivity.

closed 2024-10-08 by AlbrechtL