AlbrechtL/welle.io #558: Support for DAB SPI

nickpiggott, 2020-05-21:
Is there interest in supporting decoding of DAB SPI?

This would support:
* Reception of Station Logos
* Reception of Programme Schedules

There isn't an open source DAB EPG decoder, so this might need to be a "from scratch" activity.

AlbrechtL, 2020-05-29:
Can you explain more in detail what a DAB SPI is?

nickpiggott, 2020-05-30:
DAB SPI was previously known as DAB EPG. ([TS 102 818](https://www.etsi.org/deliver/etsi_ts/102800_102899/102818/03.01.01_60/ts_102818v030101p.pdf))

SPI is (usually) transmitted as a packet data service, one per multiplex, and contains
* Station logos (3 sizes per station)
* Programme Schedule files

DAB SPI is implemented in car receivers (BMW, Audi, etc.) and was previously integrated into receivers like the PURE Evoke-3, iRiver B20, Roberts MP-SOUND 41.

SPI can also be acquired over IP, using a RadioDNS Lookup. When acquired over IP, it usually contains:
* Station logos (5 sizes)
* Genre information
* IP streaming URLs
* Programme Schedule files

The HRadio Project has implemented an [SPI over IP package in Java](https://github.com/hradio/radiodns), which they've used in their [Android app](https://play.google.com/store/apps/details?id=lmu.hradio.hradioshowcase), and it appears to have been integrated into [DAB-Z](https://play.google.com/store/apps/details?id=com.zoulou.dab) too.

As welle.io implements DAB Slideshow (with an MOT Decoder), implementing DAB SPI would need a EPG Binary Decoder and an XML parser. Implementing SPI over IP needs an XML parser. The output XML files are identical between DAB and IP, so can be rendered using an single UI element.

AlbrechtL, 2020-06-09:
Thanks for the explanation.

As far as I know there is no EPG Binary Decoder specification public available.

Do you know if DAB SPI is already in use?

andimik, 2020-06-09:
Yes, there are lots of muxes all over Europe using SPI

The only software which can display it (but only the preview, no export of the thumbnails possible) is the Noxon Software.

Qt-DAB can export the content to xml files. But I could not find a suitable parser for them.

nickpiggott, 2020-06-09:
Ben Poor (formerly from Global Radio) has a binary encoder which gives the mappings:

https://github.com/magicbadger/python-dabepg

The also has an "unmarshall" method that I've never tried:

`
def unmarshall(i):
`
`   """Unmarshalls a PI or SI binary file to its respective :class:Epg or :class:ServiceInfo object
`    
`    :param i: String or File object to read binary from
`
`
    :type i: str, file
    """    
`

Might that help?

AlbrechtL, 2020-07-02:
The best case would be to have a C/C++ decoder that we can just integrate.

But as a first step we should consider to create files that we can put into [python-dabepg](https://github.com/magicbadger/python-dabepg).

@nickpiggott Do you know which kind of input data `python-dabepg` needs?

nickpiggott, 2020-07-03:
I'll talk with @magicbader to be certain, but I think it's the binary
payload of an MOT Object received over DAB. So if you have an MOT decoder
already that extracts the PNG/JPG images for Slideshow, you can use that

It needs to decode:
* MOT Directory - tells you which object is the SI file, which are the PI
files, and which are the logo files (as PNG or JPG)
* Extract the MOT Objects to their binary payloads
* Binary payloads of the station logos -> send to an image decoder and
cache
* BInary payloads of SI and PI -> send to python-dabpeg, which will create
an in-memory nested data structure representing the XML of SI/PI
* Traverse the data structure to associate the logo files with services,
and extract things like station descriptions etc.

Nick

andimik, 2020-07-03:
Please correct me if I am wrong, but Qt-DAB also can handle it.

The only problem there is that I couldn't find a program which display the xml files.

But Qt-DAB can extract the slides (not 100% accurate, does not work on all muxes) and the content.

nickpiggott, 2020-07-03:
I think Qt-DAB may only support Slideshow. SPI is a different application,
but both of them use MOT as a transport protocol.

SPI requires MOT Directory to be supported (which Slideshow does not), and
SPI consists of binary-encoded XMI files as well as image files of
differing resolutions (usually representing station logos).

So it's important not to confuse Slideshow with SPI, even though they both
use MOT as transport.

andimik, 2020-07-03:
This is what you get from Qt-DAB when you for instance use SWR Mux (feed 7 deg East)

```
handle_MOT: type 700, name ee010ea.EIB dir = 1
handle_MOT: type 203, name radio_swr-dasding_128x128v.png dir = 1
going to write file /tmp/qt-pictures/radio_swr-dasding_128x128v.png
handle_MOT: type 203, name radio_swr-swr-aktuell_128x128v.png dir = 1
going to write file /tmp/qt-pictures/radio_swr-swr-aktuell_128x128v.png
handle_MOT: type 203, name radio_swr-swr1_128x128v.png dir = 1
going to write file /tmp/qt-pictures/radio_swr-swr1_128x128v.png
handle_MOT: type 203, name radio_swr-swr2_128x128v.png dir = 1
going to write file /tmp/qt-pictures/radio_swr-swr2_128x128v.png
handle_MOT: type 203, name radio_swr-swr3_128x128v.png dir = 1
going to write file /tmp/qt-pictures/radio_swr-swr3_128x128v.png
handle_MOT: type 203, name radio_swr-swr4_128x128v.png dir = 1
going to write file /tmp/qt-pictures/radio_swr-swr4_128x128v.png
handle_MOT: type 701, name w20200325dd301c0.EHB dir = 1
handle_MOT: type 701, name w20200325dd3a2c0.EHB dir = 1
handle_MOT: type 701, name w20200325dd3a3c0.EHB dir = 1
handle_MOT: type 701, name w20200325dd3a5c0.EHB dir = 1
handle_MOT: type 701, name w20200325dd3a6c0.EHB dir = 1
handle_MOT: type 701, name w20200325dd804c0.EHB dir = 1
handle_MOT: type 701, name w20200325dd904c0.EHB dir = 1
handle_MOT: type 701, name w20200325ddc04c0.EHB dir = 1
handle_MOT: type 701, name w20200325ddd04c0.EHB dir = 1
```
[image]
[image]
[image]
[image]
[image]
[image]

The newer versions don't write the EHB files any more.

This was what I got 2 year ago.

[epg.zip](https://github.com/AlbrechtL/welle.io/files/4870929/epg.zip)

For instance 

```
<epg system="DAB">
 <schedule version="0" creationTime="2017-07-28T15:00:06Z">
  <scope startTime="2017-07-29T00:00:00+02:00" stopTime="2017-07-30T00:00:00+02:00">
   <serviceScope id="40.e0.10"/>
  </scope>
  <programme xml:lang="de" version="0" broadcast="on-air" recommendation="no" shortId="817967">
   <mediumName xml:lang="de">Soundtrack</mediumName>
   <longName xml:lang="de">Soundtrack</longName>
   <location>
    <time duration="PT7H" time="2017-07-29T00:00:00+02:00"/>
   </location>
   <mediaDescription>
    <shortDescription xml:lang="de">00:00-00:03 Nachrichten

06:00-06:03 Nachrichten</shortDescription>
    <longDescription xml:lang="de">

00:00-00:03 Nachrichten

06:00-06:03 Nachrichten</longDescription>
   </mediaDescription>
   <genre type="main" href="urn:tva:metadata:cs:ContentCS:2005:3.6"/>
   <genre type="main" href="urn:tva:metadata:cs:ContentCS:2005:3.6.4.1"/>
  </programme>
  <programme xml:lang="de" version="0" broadcast="on-air" recommendation="no" shortId="817970">
   <mediumName xml:lang="de">Early Bird</mediumName>
   <longName xml:lang="de">Early Bird</longName>
   <location>
[... 106 lines cut ...]

21:00-21:03 Nachrichten

22:00-22:03 Nachrichten

23:00-23:03 Nachrichten</shortDescription>
    <longDescription xml:lang="de">

19:00-19:03 Nachrichten

20:00-20:03 Nachrichten

21:00-21:03 Nachrichten

22:00-22:03 Nachrichten

23:00-23:03 Nachrichten</longDescription>
   </mediaDescription>
   <genre type="main" href="urn:tva:metadata:cs:ContentCS:2005:3.6"/>
  </programme>
 </schedule>
</epg>
```

I'm not a programmer, but I guess that's what you meant.

andimik, 2020-07-03:
Can we reuse this files?

https://github.com/MichalLeonBorsuk/drm-code/tree/622cc259e15f25ec5432ec3aa8176147c0a7f94d/branches/dream-980/src/datadecoding

AlbrechtL, 2020-07-11:
@nickpiggott Thanks for your explanation.

I checked an exemplary audio service. There the slideshow is coming via X-PAD and the MOT protocol. But there is no MOT dictionary included. Thus I assume that the MOT dictionary is transmitted inside a data service. But how do I know which is the corresponding data service to an audio service? Inside the my ensemble there are multiple data services.

nickpiggott, 2020-07-11:
@AlbrechtL 

I will refer to EPG as SPI, as SPI is the new name for the application.

The SPI service will always use MOT, and it may be transmitted:
* In the X-PAD of the radio service it relates to _(Not very common)_
AND/OR
* As an MSC Packet Data service carrying information on one or more radio services _(Commonly implemented)_

**X-PAD - with Slideshow and SPI**

There will be ONE FIG0/13 entry referencing the current audio service (SId / SCId). It will have "Number of User Applications = 2"

- User Application 1 = 0x02 (Slideshow), X-PAD AppTy = X* (MOT), DSCTy = 60 (MOT)
- User Application 2 = 0x07 (EPG), X-PAD AppTy = Y* (MOT), DSCTy = 60 (MOT) 

(* this value is created by the encoder - typically 12 for Slideshow, 16 for EPG)

When decoding X-PAD (EN 300 401 s 7.4.2.0 and EN 301 234 s 5.2.2) the Content Indicator flag will be set (CI Flag = 0b1), and each X-PAD field will start with at least TWO 1 byte Contents Indicators (s 7.4.4):

- X-PAD AppTy = 1 (MOT Data Group Length Indicator)
- X-PAD AppTy = X or X+1 (Slideshow MOT Data Group Start / Continuation) 
OR
- X-PAD AppTy = 1 (MOT Data Group Length Indicator)
- X-PAD AppTy = Y or Y+1 (EPG MOT Data Group Start / Continuation)

(* this value as advertised in FIG0/13 above)

Use the same process to rebuild Data Groups from the X-PAD fields as you do for Slideshow, but keep the two applications (X and Y) separate. Each complete Data Group creates an MOT Segment, and a complete MOT Object is rebuilt from one or more MOT Segments. (EN 301 234 s5.1).

**MSC Packet Data Channel with SPI**

You need to use FOUR FIGs to locate an SPI service:

- FIG0/2 with P/D = 1, TMId = 0b11 (MSC Packet Data Channel). Provides:  32bit SId and a 12bit SCId.
- FIG0/3 with P/D =1 , SCId (matching FIG0/2), DSCTy = 60 (MOT). Provides: SubChId, Packet Address
- FIG0/8 with P/D = 1, 32bit SId, 12 bit SCId (which match FIG0/2). Provides: 4 bit SCIdS value.
- FIG0/13 with P/D =1, 32bit SId (matching FIG0/2), and SCIds (matching FIG0/8). Provides: User Application Type

So locate the SubChID that has TMId= 11 (Packet Data), DSCTy = 60 (MOT), User Application Type 0x07 (SPI). Extract the packets from that SubChID that match Packet Address. 

Concatenate those packets to make MSC Data Groups -> MOT Segments -> MOT Objects

Currently you can't tell which services are included in an SPI service without decoding it. (There is a debate about how to signal this, such as making the SPI MSC Packet Data service a component of the audio service, but this isn't established). So it's usual to just decode the SPI service(s) and cache them on the receiver.

**Common Process**

When decoding SPI, you will receive MOT Bodies and ONE MOT Directory. The MOT Directory contains the headers for all the MOT Bodies (referenced by their Transport ID). Therefore you can't decode any of the MOT Objects until you have the MOT Directory.

- MOT Directory is transported in MSC Data Group type 6
- Compressed MOT Directory is transported in MSC Data Group type 7 - you have to pass the payload through GZIP to expand it.
- MOT Bodies are transported in MSC Data Group type 4.

Once you've decoded the MOT Objects, you will have:
* ONE SI file - Content Type = 7 and Content SubType = 0
* ZERO or more PI files - ContentType = 7 and Content SubType =1
* ZERO or more other files (representing the logos usually) - with content types per EN 301 234 / TS 101 757 Table 17.

Final step - you need to binary decode the SI and PI files, which you can push through a binary decoder, that will create SI.xml and PI.xml files. The SI has the service information, and the links to logos will refer to ContentName values from the MOT Objects. The PI files have programme schedule information, usually one file per service per calendar day.

**EXAMPLE ETI FILE**
This [ETI file](https://drive.google.com/file/d/1HmhAfJQAs9Pf70cjPIpy87MtM8LMNg8q/view?usp=sharing) contains a snapshot of the Small Scale DAB Trial in Bristol UK (taken 2020-07-11 at 19:20).

There is a MSC Packet Data service with SId = e1c000d4. It's an 8kbit/s channel, so there is 1 x 24 byte packet per Common Interleaved Frame. The Packet Address is 0x01.

The SPI service contains:
- 1 x SI file (20200711_e1.c1c8.SI.xml)
- 6 x PI files (3 days of Programme Information for each of services c3bb and cf92)
- 36 x PNG files (4 logos for each of 9 services)

(Not all services in the ensemble are represented in the SI file).

andimik, 2020-07-12:
Thanks for the file, just fyi: Qt-DAB does not find anything in the SPI service (not even a png file), so you cannot rely on Qt-DAB's code here.

nickpiggott, 2020-07-12:
My apologies @andimik - that ETI file was corrupted. I've uploaded a new version [here]( https://drive.google.com/file/d/1YKE8elPkc4_0nzejYfJSvEZ3G2RTwAZy/view?usp=sharing) which is only 176MB long, and I've tested with DAB-Scout 3 to be working.

The MOT Directory appears close to the end of the file, so it had to loop in DAB scout to be able to decode everything.

Could you try again with Qt-DAB please?

andimik, 2020-07-12:
Are you sure this file is NOT corrupt? I even get no audio in the today's file whereas yesterday it worked.

Have checked it with dablin_gtk.

nickpiggott, 2020-07-12:
I recorded that file from the test multiplexer and didn't configure any
audio inputs. It did decode correctly in DAB-Scout 3.

You should see a MSC Packet Data channel with SID e1c000d4, which has the
SPI application in it (packet address 1). If you're looking at FIGs, 0/2,
0/3, 0/8 and 0/13 should all reference to it.

Nick

andimik, 2020-07-12:
Oh I see, I thought it was a live capture ...

I get this for example 

`FIG: 0/3
     Length: 6
     OE: 0
     C/N: 0
     P/D: 0
     Decoding:
      -
       SCId: 0x0
       CAOrg flag: 0 CAOrg field absent
       DG flag: 0
       data groups are used to transport the service component
       DSCTy: 60 Multimedia Object Transfer (MOT)
       SubChId: 0xE
       Packet address: 0x1
`
`   Stream Number:
    value: 13
   SCID:
    value: 14
    desc: Sub-channel Identifier
   SAD:
    value: 828
    desc: Sub-channel Start Address
   TPL:
    desc: Sub-channel Type and Protection Level
    EEP:
     value: 34
     desc: Equal Error Protection
     Level: 3-A
     Rate: 1/2
     CUs: 6
   STL:
    value: 3
    desc: Sub-channel Stream Length
   bitrate:
    value: 8
    desc: kbit/s
`

I can see the service "EPG" in all my tools, but the Fraunhofer Software is running for lots of minutes without showing the SPI content.

[image]

Qt-DAB says
```
Data Services
program name;;serviceId;subchannelId;start address;length (CU); bit rate; FEC; prot level; appType ; subService ; 

EPG             ;United Kingdom;E1C000D4;14;828;6;8;0;EEP 3-A;7;no;;
```

That's why I even fear this new ETI is corrupt.

nickpiggott, 2020-07-12:
Hello

That signalling looks correct. I am able to decode the file using DAB Scout.

The SPI uses about 22,000 frames to receive, so probably about 10-12
minutes. DAB Scout is looping the file to get all the objects because the
MOT Directory comes last. Does the Fraunhofer decode indicate that it's
found any MOT objects? An MOT Directory?

I'll make a file twice as long, so the SPI loops twice and see if that
works.

Nick

andimik, 2020-07-12:
Yes, please make it much longer.

andimik, 2020-07-12:
BTW: Don't use your mail tool, please log on to github as you missed my updated comments.

andimik, 2020-07-12:
Does your license allow you to increase the bitrate? There are 30 CUs free.

nickpiggott, 2020-07-12:
I'm trying not to change too many variables, and I'm not going to change the configuration of the on-air service.

In [this recording](https://drive.google.com/file/d/1trChjPd6jzdecNzgD4m1uboGl2TxnShv/view?usp=sharing) I've halved the payload size by omitting the 320x240px logos - it should not be received in just over 8,000 frames (3.5 minutes), and there are at least three full rotations of the carousel in the file.

andimik, 2020-07-12:
Now it loads it completely, but there is no preview in the next week available. Should I find events in this SPI?

[image]

At least I find these pointers in the Fraunhofer Software (hidden folder `.mmp\cache\spi` in user directory)

```
$ cat /media/andreas/DOS/db.mmp 
MMP-STORAGE�_
�����������������	��_
������
      ����
          Sam FMSam FM BristolSam FM Bristol����"Sam_FM_112_32.pngimage/png Sam_FM_32_32.pngimage/png$Sam_FM_128_128.pngimage/png��ɇ_
������
      ��ɇ
         BStoke�Bradley Stoke��������"BStoke_112_32.pngimage/png BStoke_32_32.pngimage/png$BStoke_128_128.pngimage/png����_
������
      ���BCfBCfm��������BCfm_112_32.pngimage/pngBCfm_32_32.pngimage/png BCfm_128_128.pngimage/png����_
������
      ����
NOODSNOODS Radio�������� NOODS_112_32.pngimage/pngNOODS_32_32.pngimage/png"NOODS_128_128.pngimage/png����_
������
      ����BFBS GrkBFBS Gurkha��������&BFBS_Grk_112_32.pngimage/png$BFBS_Grk_32_32.pngimage/png(BFBS_Grk_128_128.pngimage/png����_
������
      ����HUBHub Radio��������HUB_112_32.pngimage/png�HUB_32_32.pngimage/pngHUB_128_128.pngimage/png��û_
������
      ��û
         BreezeBreeze Bristol$The Breeze Bristol����"Breeze_112_32.pngimage/png Breeze_32_32.pngimage/png$Breeze_128_128.pngimage/png����_
������
      ����
          SomerVSomer Valley��������"SomerV_112_32.pngimage/png SomerV_32_32.pngimage/png$SomerV_128_128.pngimage/png����_
������
      ����
Ujima
Ujima�������� Ujima_112_32.pngimage/pngUjima_32_32.pngimage/png"Ujima_128_128.pngimage/png��
```

but I could not see any png so far.

nickpiggott, 2020-07-12:
That db.mmp looks like the contents of the SPI SI.xml file that has been decoded - you can see the references to the station names and logo files and so on.

There should be programme schedules for THREE days (12th, 13th, 14th July 2020) for:

* The Breeze (c3bb)
* Sam FM (cf92)

There should be logos at 32 x 32, 112 x 32 and 128 x 128 pixels for
* The Breeze
* Sam
* Bradley Stoke Radio
* BCfm
* Ujima Radio
* NOODS
* Hub Radio
* BFBS Gurkha Radio
* Somer Valley Radio

All those are decoding correctly in DAB Scout - I can see programme schedules, and I can see the logos correctly decoded in the cache file. (DAB Scout 3 doesn't show the logos in the UI).

Is there any insight into what the Fraunhofer analyser doesn't like? It can clearly decode the MOT Objects as it has decoded the MOT Directory at the very least.

nickpiggott, 2020-07-12:
From DAB Scout 3

[image]

and the data directory

[image]

and I've attached the decoding of the MOT Directory which DAB Scout reformats to XML

[_motDirectory.xml.txt](https://github.com/AlbrechtL/welle.io/files/4908668/_motDirectory.xml.txt)

AlbrechtL, 2020-07-12:
> I will refer to EPG as SPI, as SPI is the new name for the application.
>
> The SPI service will always use MOT, and it may be transmitted:
>
>  * In the X-PAD of the radio service it relates to (Not very [...]
Thanks for the very detailed description! To me it looks like that this is very interesting to implement.

andimik, 2020-11-08:
Dear Albrecht and Nick,

Jan has implemented a simple SPI decoder (still in progress!!!) and can show the SPI in a separate window.

nickpiggott, 2020-11-08:
That's great news! Is there a build ready to test with, and I can check it out with some UK / Bristol services I know are broadcasting SPI?

andimik, 2020-11-08:
I forgot to mention that it's inside Qt-DAB

First of all add these lines to `~/.qt-dab.ini`

```
epgPath=/tmp/qt-dab/epg/
filePath=/tmp/qt-dab/mot/
```

Then you have to select the SPI service and wait till the EHB files have been written (check the console for messages)

For example 

```
starting a backend for EPG              (E0D010C9)
handle_MOT: type 701 (7), name w20200411dd804c0.EHB dir = 1
epg
currentSID = D804
Version 1
epg file 
handle_MOT: type 701 (7), name w20200411dd904c0.EHB dir = 1
epg
currentSID = D904
Version 1
epg file 
```

Then you select one service (which has a preview) and open the detail window.

There you should see a new button "timeTable".

https://www.rundfunkforum.de/download/file.php?id=5419&mode=view

AlbrechtL, 2020-11-21:
Cool! Thanks for the hint. Since welle.io bases on Jan codes the integration of Jans SPI decoder shouldn't difficult.

andimik, 2021-11-28:
Hint: cQIRX (Qirx Console) can write all slides and has an XML Parser for the EIB File. 

https://qirx.softsyst.com/LinuxCmdLineInstall

But it's not Open Source. 

The Linux version allows to select even data services which are not detected as SPI (Like in Slovenia, where the EPG/SPI Service is called Spored).

The Windows Version works with most of the available muxes. This is the only software (besides of Fraunhofer) that uses 32x32 thumbnails for the audio services.

nickpiggott, 2021-12-01:
Thanks for that suggestion @andimik - I really find it useful to see what's being broadcast on SPI.

Why is the Slovenian service not being detected as SPI? If it's being signalled as SPI (DAB EPG) application, it shouldn't matter what the service label is?

andimik, 2021-12-01:
Without having read the standard, Jan and Clem told me they have to identify SPI by their service label names.

Is there really a flag which could help identifying SPI?

nickpiggott, 2021-12-01:
I did a [long walk-through](https://github.com/AlbrechtL/welle.io/issues/558#issuecomment-657121671) of how you "unthread" the configuration of data services from their various FIGs. 

In summary, it's identified from the User Application Type transmitted in FIG0/13, which should have a value 0x07 for SPI. If it's not being transmitted with that, it's not compliant with the specification, and its unlikely any compliant receiver will identify and decode it.

(There were, back in the year 20xx, some receivers that speculatively decoded packet data services with MOT Directory, to see if they contained files ending EIB. But that was never part of the standard, and is not at all reliable. The current standard, for example, does not require objects to be named ending EIB - they're identified by their MOT Headers).

AlbrechtL, 2022-01-06:
Stupid question: Has Jan with Qt-DAB some progress in SPI decoding?

andimik, 2022-01-07:
No, although there are lots of progresses regarding other features, the SPI is still in same experimental stage...

It even does not show the date, just the time.

AlbrechtL, 2022-02-16:
Does anymore knows how Andreas Gsinn [DAB Player](https://www.ukwtv.de/cms/downloads-aside/281-dab-player-von-andreas-gsinn.html) is getting the SPI? Does he has its own implementation?

andimik, 2022-02-17:
On this German page he writes, this is in preparation, but the same text is online for a very long time now.

There is no support for SPI in DAB Player.

https://www.ukwtv.de/cms/downloads-aside/281-dab-player-von-andreas-gsinn.html

AlbrechtL, 2022-02-17:
Oh, you are right. I thought he implemented it but I mixed up something, obviously.

AlbrechtL, 2022-03-14:
> @AlbrechtL
> 
> I will refer to EPG as SPI, as SPI is the new name for the application.
> 
> The SPI service will always use MOT, and it may be transmitted:
> 
>     * In the X-PAD of the radio service it relates to _(Not very common)_
>       AND/OR
> 
>     * As an MSC Packet Data service carrying information on one or more radio services _(Commonly implemented)_
> 
> 
> **X-PAD - with Slideshow and SPI**
> 
> There will be ONE FIG0/13 entry referencing the current audio service (SId / SCId). It will have "Number of User Applications = 2"
> 
>     * User Application 1 = 0x02 (Slideshow), X-PAD AppTy = X* (MOT), DSCTy = 60 (MOT)
> 
>     * User Application 2 = 0x07 (EPG), X-PAD AppTy = Y* (MOT), DSCTy = 60 (MOT)
> 
> 
> (* this value is created by the encoder - typically 12 for Slideshow, 16 for EPG)
> 
> When decoding X-PAD (EN 300 401 s 7.4.2.0 and EN 301 234 s 5.2.2) the Content Indicator flag will be set (CI Flag = 0b1), and each X-PAD field will start with at least TWO 1 byte Contents Indicators (s 7.4.4):
> 
>     * X-PAD AppTy = 1 (MOT Data Group Length Indicator)
> 
>     * X-PAD AppTy = X or X+1 (Slideshow MOT Data Group Start / Continuation)
>       OR
> 
>     * X-PAD AppTy = 1 (MOT Data Group Length Indicator)
> 
>     * X-PAD AppTy = Y or Y+1 (EPG MOT Data Group Start / Continuation)
> 
> 
> (* this value as advertised in FIG0/13 above)
> 
> Use the same process to rebuild Data Groups from the X-PAD fields as you do for Slideshow, but keep the two applications (X and Y) separate. Each complete Data Group creates an MOT Segment, and a complete MOT Object is rebuilt from one or more MOT Segments. (EN 301 234 s5.1).
> 
> **MSC Packet Data Channel with SPI**
> 
> You need to use FOUR FIGs to locate an SPI service:
> 
>     * FIG0/2 with P/D = 1, TMId = 0b11 (MSC Packet Data Channel). Provides:  32bit SId and a 12bit SCId.
> 
>     * FIG0/3 with P/D =1 , SCId (matching FIG0/2), DSCTy = 60 (MOT). Provides: SubChId, Packet Address
> 
>     * FIG0/8 with P/D = 1, 32bit SId, 12 bit SCId (which match FIG0/2). Provides: 4 bit SCIdS value.
> 
>     * FIG0/13 with P/D =1, 32bit SId (matching FIG0/2), and SCIds (matching FIG0/8). Provides: User Application Type
> 
> 
> So locate the SubChID that has TMId= 11 (Packet Data), DSCTy = 60 (MOT), User Application Type 0x07 (SPI). Extract the packets from that SubChID that match Packet Address.
> 
> Concatenate those packets to make MSC Data Groups -> MOT Segments -> MOT Objects
> 
> Currently you can't tell which services are included in an SPI service without decoding it. (There is a debate about how to signal this, such as making the SPI MSC Packet Data service a component of the audio service, but this isn't established). So it's usual to just decode the SPI service(s) and cache them on the receiver.
> 
> **Common Process**
> 
> When decoding SPI, you will receive MOT Bodies and ONE MOT Directory. The MOT Directory contains the headers for all the MOT Bodies (referenced by their Transport ID). Therefore you can't decode any of the MOT Objects until you have the MOT Directory.
> 
>     * MOT Directory is transported in MSC Data Group type 6
> 
>     * Compressed MOT Directory is transported in MSC Data Group type 7 - you have to pass the payload through GZIP to expand it.
> 
>     * MOT Bodies are transported in MSC Data Group type 4.
> 
> 
> Once you've decoded the MOT Objects, you will have:
> 
>     * ONE SI file - Content Type = 7 and Content SubType = 0
> 
>     * ZERO or more PI files - ContentType = 7 and Content SubType =1
> 
>     * ZERO or more other files (representing the logos usually) - with content types per EN 301 234 / TS 101 757 Table 17.
> 
> 
> Final step - you need to binary decode the SI and PI files, which you can push through a binary decoder, that will create SI.xml and PI.xml files. The SI has the service information, and the links to logos will refer to ContentName values from the MOT Objects. The PI files have programme schedule information, usually one file per service per calendar day.
> 
> **EXAMPLE ETI FILE** This [ETI file](https://drive.google.com/file/d/1YKE8elPkc4_0nzejYfJSvEZ3G2RTwAZy/view?usp=sharing) contains a snapshot of the Small Scale DAB Trial in Bristol UK (taken 2020-07-12 at 09:50 UTC - Updated to replace previously corrupted file).
> 
> There is a MSC Packet Data service with SId = e1c000d4. It's an 8kbit/s channel, so there is 1 x 24 byte packet per Common Interleaved Frame. The Packet Address is 0x01.
> 
> The SPI service contains:
> 
>     * 1 x SI file (20200711_e1.c1c8.SI.xml)
> 
>     * 6 x PI files (3 days of Programme Information for each of services c3bb and cf92)
> 
>     * 36 x PNG files (4 logos for each of 9 services)
> 
> 
> The total SPI service spans 20,216 packets, so it takes just over 8 minutes to decode normally.
> 
> (Not all services in the ensemble are represented in the SI file).

@nickpiggott Today I started in taking a look into it. So my first question is: How do I know which services is carrying the SPI data? In my case I have 5 data services. I took a look into the DAB standards but I didn't find a corresponding data service announcement or signal.
```
2022-03-14T21:18:56.426 Debug: RadioController: Found data service  E0D110BC "EPG Deutschland "
2022-03-14T21:18:56.471 Debug: RadioController: Found data service  E0D010BC "TPEG            "
2022-03-14T21:18:56.506 Debug: RadioController: Found data service  E0D310BC "TPEG_MM         "
2022-03-14T21:18:56.554 Debug: RadioController: Found data service  E0D210BC "PPP-RTK-AdV     "
2022-03-14T21:18:56.787 Debug: RadioController: Found data service  E0D00250 "DRadio Daten    "
```

AlbrechtL, 2022-03-15:
Thanks. This makes it a bit more clear what is actual to do.

Now, I realized that I may didn't say clearly what my current issue is.

I have a number of audio services which may not contain SPI information.
```
2022-03-15T19:48:59.238 Debug: RadioController: Found service  D210 "Dlf             "
2022-03-15T19:48:59.638 Debug: RadioController: Found service  1A64 "ERF Plus        "
2022-03-15T19:48:59.921 Debug: RadioController: Found service  D220 "Dlf Kultur      "
2022-03-15T19:49:00.004 Debug: RadioController: Found service  D240 "DRadio DokDeb   "
2022-03-15T19:49:00.157 Debug: RadioController: Found service  D75B "KLASSIK RADIO   "
2022-03-15T19:49:00.207 Debug: RadioController: Found service  15DC "sunshine live   "
2022-03-15T19:49:00.238 Debug: RadioController: Found service  D01C "radio horeb     "
2022-03-15T19:49:00.288 Debug: RadioController: Found service  100D "Schwarzwaldradio"
2022-03-15T19:49:00.321 Debug: RadioController: Found service  1A45 "ENERGY          "
2022-03-15T19:49:00.358 Debug: RadioController: Found service  10C3 "SCHLAGERPARADIES"
2022-03-15T19:49:00.404 Debug: RadioController: Found service  17FA "Absolut relax   "
2022-03-15T19:49:00.438 Debug: RadioController: Found service  15DD "RADIO BOB!      "
2022-03-15T19:49:00.487 Debug: RadioController: Found service  D230 "Dlf Nova        "
```
I expected to have a link from audio service to a SPI contained data service. Just guessing from the name `D210 "Dlf             " `can be related with E0D00250 `"DRadio Daten    "`. But where is the link? 

Data services:
```

2022-03-14T21:18:56.426 Debug: RadioController: Found data service  E0D110BC "EPG Deutschland "
2022-03-14T21:18:56.471 Debug: RadioController: Found data service  E0D010BC "TPEG            "
2022-03-14T21:18:56.506 Debug: RadioController: Found data service  E0D310BC "TPEG_MM         "
2022-03-14T21:18:56.554 Debug: RadioController: Found data service  E0D210BC "PPP-RTK-AdV     "
2022-03-14T21:18:56.787 Debug: RadioController: Found data service  E0D00250 "DRadio Daten    "
```

Or is the case in just looking to every data service for SPI?

In my first step I wanted to get station images because we already have a MOT decoder.

AlbrechtL, 2024-10-09:
I'm closing this issue because the DAB data services are not in the focus right now.

If somebody provides a PR, that would be fantastic! To have station logs would be really nice.

AlbrechtL, 2025-03-11:
I just saw that qt-dab (https://github.com/JvanKatwijk/qt-dab) implements an EPG now including stations logos now. Since welle.io is a fork of qt-dab (forked 8 years ago) it should not too difficult to use Jan's code.

For that reason I reopen this issue.

andimik, 2025-03-11:
Hint: https://github.com/KejPi/AbracaDABra can handle it, too.

AlbrechtL, 2025-03-12:
Yes, but AbracaDABra is not open source as far as I saw.

AlbrechtL, 2026-10-11:
Thanks to AI welle.io supports stations logos now.

[image]

See the next branch. CI builds can be found here: https://welle-io-nightlies.albrechtloh.de/

AlbrechtL, 2026-10-22:
Done.

closed 2026-10-22 by AlbrechtL

andimik, 2026-10-22:
Yes, very nice!

[image]