AlbrechtL/welle.io #559: Allow switching channel on get request

athoik, 2020-05-24:
Hi, 

We need to have stream channels on enigma2 receiver (something like using m3u list).
Everything works fine, assuming that only one channel involve (using url /mp3/SID)
If more than channel involved, we need a way to change frequency  (without having user to open the welle-cli web interface).

The following patch does the job, although most probably in a hacky way. 

It allows to specify the accepted channels on url, when channel is different it will retune.
Url to use that feature is /mp3/CHANNEL/SID eg /mp3/12C/0x1001.

```
@@ -474,9 +474,38 @@ bool WebRadioInterface::dispatch_client(Socket&& client)
                 const regex regex_slide(R"(^[/]slide[/]([^ ]+))");
                 std::smatch match_slide;

+                const regex regex_tune_mp3(R"(^[/]mp3[/]([56789][A-D]|1[123][A-D]|13[EF]|L[A-P])[/]([^ ]+))");
                 const regex regex_mp3(R"(^[/]mp3[/]([^ ]+))");
                 std::smatch match_mp3;
-                if (regex_search(req.url, match_mp3, regex_mp3)) {
+                if (regex_search(req.url, match_mp3, regex_tune_mp3)) {
+                    cerr << "GET channel: " << match_mp3[1] << endl;
+                    try {
+                        const auto freq = input.getFrequency();
+                        const auto chan = channels.getChannelForFrequency(freq);
+                        const auto newchan = match_mp3[1];
+                        if (newchan != chan) {
+                            retune(newchan);
+                            this_thread::sleep_for(chrono::seconds(2));
+                            success = send_mp3(s, match_mp3[2]);
+                            for (auto i=0; i<10 && !success; i++) {
+                                cerr << "Not yet synced, delaying to reply..." << endl;
+                               success = send_mp3(s, match_mp3[2]);
+                                this_thread::sleep_for(chrono::seconds(2));
+                            }
+                            if (success) {
+                                return true;
+                            }
+                        }
+                        else {
+                            cerr << "Retune not required..." << endl;
+                        }
+                    }
+                    catch (const out_of_range& e) {
+                        cerr << "Failed to retune " << e.what() << endl;
+                    }
+                    success = send_mp3(s, match_mp3[2]);
+                }
+                else if (regex_search(req.url, match_mp3, regex_mp3)) {
                     success = send_mp3(s, match_mp3[1]);
                 }
                 else if (regex_search(req.url, match_slide, regex_slide)) {
```

If above is acceptable, I can create a proper PR.

Thanks

mpbraendli, 2020-05-26:
Yes it feels a bit hackish. I understand the idea but I don't like the approach so much.

athoik, 2020-05-26:
Another possibility is to allow only get channel /channel/12C that allows to retune, followed by normal /mp3/sid.

It will allow easier change.

I know it's looks and feels hackish. Just sharing the idea ;)

AlbrechtL, 2020-07-12:
Any news here?

FdeAlexa, 2020-09-18:
@athoik 
I need this change: would you please be so kind to explain how to apply this patch?
Thanks a lot and best regards,
Francesco

FdeAlexa, 2020-09-18:
Hello to everybody, I have solved my needs:
the module "WebRadioInterface::dispatch_client(Socket&& client)", that is contained in the source file "webradiointerface.cpp" has to be modified, as reported on the first message of Athanasios Oikonomou on this post. The only difference is the offset of the modification that is 496 instead of 474 for the last version of welle.cli.
Thanks and best regards,
Francesco

AlbrechtL, 2020-09-30:
@mpbraendli Does it make sense to integrate this change into welle-cli? Otherwise I would close this issue.

mpbraendli, 2020-10-01:
I don't like the approach, `GET` is not the appropriate HTTP method for switching the channel.

FdeAlexa, 2020-10-01:
Hello everybody,
Thanks for your contributions.
Matthias, your comment is appreciated but I think you could be very kind if you can suggest a different approach instead of "GET".
My knowledge of "programming" does not allow me to comment on any technicalities, but I really appreciated the idea: I integrated it into my moOde player and then I would like this change to be integrated into welle-cli.
Thank you very much and sorry for my not very good english,
Best regards,
Francesco

closed 2020-12-02 by AlbrechtL