AlbrechtL/welle.io #560: Allow streaming wav to stdout using welle-cli

athoik, 2020-05-24:
Hi,

Is there a possible way to stream only one channel on stdout using welle-cli?

Something like `welle-cli -c 12C -p 0x1001 -stdout` ?

Thanks

andimik, 2020-05-24:
Why don't you use https://github.com/JvanKatwijk/dab-cmdline for that?

athoik, 2020-05-24:
Because it seems welle-cli can write files in wav (RIFF) format.

The dab-cmdline produces raw PCM and that doesn't work if you like to stream it on enigma2 or to vlc.

AlbrechtL, 2020-06-22:
What is your use case?

athoik, 2020-06-23:
Create a pipeline and feed a player from a specific channel from command line.

As @andimik  said, dab-cmdline offer this already, but with RAW PCM samples.

AlbrechtL, 2020-07-12:
Yeah that's clear. But what is the use case to do such an complicated setup? You can either use welle.io GUI or the MP3 feature of welle-cli.

closed 2021-05-27 by AlbrechtL