AlbrechtL/welle.io #562: SoapySDR - only one gain slider

goscickiw, 2020-06-01:
I'm using welle.io with SoapySDR, and my device (SDRplay RSP1A) has separate RF and IF gain. The gain slider in welle.io settings seems to be bound to the IF gain setting, and the RF gain seems to be always at 0 (it's actually gain reduction so actual RF gain is at 100%). As a result I'm not able to set the RF gain correctly.

I'm actually able to get a signal and listen to stations, but it's not very good because the RF gain isn't on optimal setting.

Would it be possible to add the RF gain slider, or make the program automatically detect which gain settings are available and add appropriate sliders?

My current settings:
SoapySDR
Driver arguments: `driver=sdrplay`
other fields empty
Gain: Manual, value = 45

Optimal gain values in my case should be 35 for IF and 5 for RF.

I have SoapySDRPlay installed and other programs work properly with it.
My OS is Ubuntu 20.04

AlbrechtL, 2020-06-22:
Good idea. Pull request are welcome.

mpbraendli, 2020-06-23:
SoapySDR presents a `setGain()` that sets the "overall amplification" for the device. The Airspy Soapy driver should take care of setting both gains when this function is called, which might be a separate issue in the Airspy driver.

Regardless of that, presenting all gain settings in the UI might still be worthwhile.