AlbrechtL/welle.io #563: No station found

RalfJahns, 2020-06-02:
Hello,

I would like to ask for support.

device seems to be good from kernel side of view:

[    5.129476] usb 1-14: dvb_usb_v2: found a 'TerraTec NOXON DAB Stick' in warm state
[    5.161604] usb 1-14: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[    5.161612] DVB: registering new adapter (TerraTec NOXON DAB Stick)
[    5.161954] IPv6: ADDRCONF(NETDEV_UP): eno1: link is not ready
[    5.174861] i2c i2c-6: Added multiplexed i2c bus 7
[    5.174864] rtl2832 6-0010: Realtek RTL2832 successfully attached
[    5.174870] usb 1-14: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[    5.177801] fc0013: Fitipower FC0013 successfully attached.
[    5.180349] media: Linux media interface: v0.10
[    5.185653] Linux video capture interface: v2.00
[    5.193542] rtl2832_sdr rtl2832_sdr.0.auto: Registered as swradio0
[    5.193545] rtl2832_sdr rtl2832_sdr.0.auto: Realtek RTL2832 SDR attached
[    5.193546] rtl2832_sdr rtl2832_sdr.0.auto: SDR API is still slightly experimental and functionality changes may follow

But this is the output, if I start a scan:

2020-06-02T09:55:45.704 Debug: RadioController: Tune to channel "12C" -> 227.36 MHz
2020-06-02T09:55:45.729 Info: OFDM-processor: input not ok, closing down
2020-06-02T09:55:45.777 Info: End RTLSDR thread
2020-06-02T09:55:45.841 Info: Start RTLSDR thread
2020-06-02T09:55:45.971 Info: OFDM-decoder:closing down now
2020-06-02T09:55:45.972 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: CorrelatePRS fft placement: EarliestPeakWithBinning
2020-06-02T09:55:45.990 Info: OFDM-processor: reset (0)
2020-06-02T09:55:46.008 Info: OFDM-processor:start
2020-06-02T09:55:46.370 Debug: RadioController: Tune to channel "12D" -> 229.072 MHz
2020-06-02T09:55:46.390 Info: OFDM-processor: input not ok, closing down
2020-06-02T09:55:46.394 Info: End RTLSDR thread
2020-06-02T09:55:46.457 Info: Start RTLSDR thread
2020-06-02T09:55:46.588 Info: OFDM-decoder:closing down now
2020-06-02T09:55:46.588 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: CorrelatePRS fft placement: EarliestPeakWithBinning
2020-06-02T09:55:46.607 Info: OFDM-processor: reset (0)
2020-06-02T09:55:46.625 Info: OFDM-processor:start
2020-06-02T09:55:46.709 Info: ofdm-processor: Lost coarse sync (coarseCorrector: 0; fineCorrector: 0)
2020-06-02T09:55:46.797 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:55:56.531 Debug: RadioController: Tune to channel "13A" -> 230.784 MHz
2020-06-02T09:55:56.550 Info: OFDM-processor: input not ok, closing down
2020-06-02T09:55:56.579 Info: End RTLSDR thread
2020-06-02T09:55:56.642 Info: Start RTLSDR thread
2020-06-02T09:55:56.775 Info: OFDM-decoder:closing down now
2020-06-02T09:55:56.776 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: CorrelatePRS fft placement: EarliestPeakWithBinning
2020-06-02T09:55:56.794 Info: OFDM-processor: reset (0)
2020-06-02T09:55:56.813 Info: OFDM-processor:start
2020-06-02T09:55:57.172 Debug: RadioController: Tune to channel "13B" -> 232.496 MHz
2020-06-02T09:55:57.203 Info: OFDM-processor: input not ok, closing down
2020-06-02T09:55:57.246 Info: End RTLSDR thread
2020-06-02T09:55:57.309 Info: Start RTLSDR thread
2020-06-02T09:55:57.440 Info: OFDM-decoder:closing down now
2020-06-02T09:55:57.440 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: CorrelatePRS fft placement: EarliestPeakWithBinning
2020-06-02T09:55:57.458 Info: OFDM-processor: reset (0)
2020-06-02T09:55:57.476 Info: OFDM-processor:start
2020-06-02T09:55:57.839 Debug: RadioController: Tune to channel "13C" -> 234.208 MHz
2020-06-02T09:55:57.869 Info: OFDM-processor: input not ok, closing down
2020-06-02T09:55:57.913 Info: End RTLSDR thread
2020-06-02T09:55:57.975 Info: Start RTLSDR thread
2020-06-02T09:55:58.105 Info: OFDM-decoder:closing down now
2020-06-02T09:55:58.106 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: CorrelatePRS fft placement: EarliestPeakWithBinning
2020-06-02T09:55:58.124 Info: OFDM-processor: reset (0)
2020-06-02T09:55:58.142 Info: OFDM-processor:start
2020-06-02T09:55:58.505 Debug: RadioController: Tune to channel "13D" -> 235.776 MHz
2020-06-02T09:55:58.535 Info: OFDM-processor: input not ok, closing down
2020-06-02T09:55:58.578 Info: End RTLSDR thread
2020-06-02T09:55:58.641 Info: Start RTLSDR thread
2020-06-02T09:55:58.770 Info: OFDM-decoder:closing down now
2020-06-02T09:55:58.771 Info: New Receiver Options: TII: 0 disable coarse corr: 0 freqsync: CorrelatePRS fft placement: EarliestPeakWithBinning
[... 917 lines cut ...]
2020-06-02T09:59:13.682 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:13.909 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:14.610 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:15.206 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:15.258 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:15.318 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:15.428 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:15.518 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:15.568 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:16.270 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:16.418 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:16.445 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:16.797 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:16.921 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:17.503 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:17.513 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:18.047 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:18.759 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:18.993 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:19.013 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:19.575 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:19.603 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:20.251 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:20.493 Info: ofdm-processor: SyncOnPhase failed
2020-06-02T09:59:20.637 Info: ofdm-processor: SyncOnPhase failed

andimik, 2020-06-02:
Only in case you are in Prague you need to search in the L-Band.

The (very long) log itself does not help. 

Why does the scan starts at channel 12C?

andimik, 2020-06-18:
Or you switch to expert mode and only scan on that channel where your local DAB ensemble is.

Then add the spectrum view and investigate, why no signal has been found.

AlbrechtL, 2020-06-22:
@RalfJahns It seems you don't receive any signal. Please switch on the expert view and check the spectrum. You can also try different manual gains.

AlbrechtL, 2020-07-12:
Closed because of inactivity.

closed 2020-07-12 by AlbrechtL