AlbrechtL/welle.io #564: [macos] cannot launch welle-io from dmg (Library missing)

tenzap, 2020-06-03:
Hello,

Launching welle-io provided by the new dmg
 https://bintray.com/package/files/albrechtl/welle.io/welle.io_nightly?order=asc&sort=name&basePath=build%2Fsrc%2Fwelle-gui&tab=files

results in this error:
```
Process:               welle-io [487]
Path:                  /Volumes/VOLUME/welle-io.app/Contents/MacOS/welle-io
Identifier:            com.yourcompany.welle-io
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           welle-io [487]
User ID:               501

Date/Time:             2020-06-03 13:02:59.828 +0200
OS Version:            Mac OS X 10.13.6 (17G65)
Report Version:        12
Anonymous UUID:        B0849E54-90F1-B5D4-D743-9E2CD6786937

Time Awake Since Boot: 260 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/librtlsdr.0.dylib
  Referenced from: /Volumes/VOLUME/welle-io.app/Contents/MacOS/welle-io
  Reason: image not found

Binary Images:
       0x1084dd000 -        0x108591ff3 +com.yourcompany.welle-io (0) <64EF78AA-A74C-3156-8FC9-6B63A26F3E78> /Volumes/VOLUME/welle-io.app/Contents/MacOS/welle-io
       0x1085c4000 -        0x108793ff7 +libfftw3f.3.dylib (0) <83DC22F8-76E0-3E14-A017-050146328500> /Volumes/VOLUME/welle-io.app/Contents/Frameworks/libfftw3f.3.dylib
       0x1087b6000 -        0x1087c6ff7 +libusb-1.0.0.dylib (0) <0FEDED27-A216-3523-9640-0AA349576646> /Volumes/VOLUME/welle-io.app/Contents/Frameworks/libusb-1.0.0.dylib
       0x1087cd000 -        0x108802fff +libfaad.2.dylib (0) <EAD72D70-2445-3427-A41B-0F0838FAA6B2> /Volumes/VOLUME/welle-io.app/Contents/Frameworks/libfaad.2.dylib
       0x108809000 -        0x108830fff +libmpg123.0.dylib (0) <D8D82121-F764-38B7-B463-9A1A88A1AB7C> /Volumes/VOLUME/welle-io.app/Contents/Frameworks/libmpg123.0.dylib
       0x108846000 -        0x10887cffb +libmp3lame.0.dylib (0) <1A3FD3F1-449E-30E8-BB96-D841D578F9D3> /Volumes/VOLUME/welle-io.app/Contents/Frameworks/libmp3lame.0.dylib
       0x1088b2000 -        0x1088b7ff7 +libairspy.0.dylib (0) <4FFBF2AE-71C9-3487-B04E-F73454CEEAA1> /Volumes/VOLUME/welle-io.app/Contents/Frameworks/libairspy.0.dylib
       0x114e4c000 -        0x114e96acf  dyld (551.4) <8A72DE9C-A136-3506-AA02-4BA2B82DCAF3> /usr/lib/dyld
    0x7fff58cd2000 -     0x7fff58cd3ffb  libSystem.B.dylib (1252.50.4) <CD555F3B-FDDB-35E5-A2FB-FBBF3D62031A> /usr/lib/libSystem.B.dylib

```

AlbrechtL, 2020-06-03:
@orryverducci Can you have a look into it?

orryverducci, 2020-06-07:
Thanks for the heads up. QT's deployment processes missed out librtlsdr, and I missed it when I checked.

I've created a pull request (#565) which fixes this.

AlbrechtL, 2020-06-09:
#565 is merged. Please try it again.

AlbrechtL, 2020-06-22:
Any test results here?

tenzap, 2020-06-23:
It does launch.

On another subject what are your thoughts & intentions on PR #532 (& #543 & #550)?

AlbrechtL, 2020-07-11:
Thanks for testing. Your other PRs I will review after the 2.2 version release.

PR #565 (2020-06-09): DMG build improvements
This makes a few small improvements to the macOS DMG build:

- Manually bundles librtlsdr. This fixes #564.
- Checks the binaries link to the correct paths for the libraries bundled with the app, and fixes them where they don't.
- Adds a shortcut to the Applications folder in the DMG, as is standard and expected by users.
- Tweaks the application name so it appears as 'welle.io' instead of 'welle-io'.
- Moves bintray preparation from the after_success build step to before_deploy build step. This removes duplication.
closed 2020-07-11 by AlbrechtL