AlbrechtL/welle.io #566: Ensemble ID

G4JFS, 2020-06-15:
How can I view the Ensemble Id? I can't find it in any of the windows. Service Ids are all there but I can't see the EId anywhere.

andimik, 2020-06-18:
Why do you need that?

You find it in the console.

> Debug: RadioController: ID of ensemble

Please note it's in decimal, not in hex.

G4JFS, 2020-06-18:
I can't see it, even in decimal. Here is a screenshot. Where should I look?

andimik, 2020-06-19:
You forgot to send the screenshot.

Set to expert mode and add the console window check there.

But why do you need it?

AlbrechtL, 2020-06-22:
@G4JFS Is the questions answered?

G4JFS, 2020-06-22:
I have the console window open but don't see the EId there.
[image]

andimik, 2020-06-23:
Check again and look for the text I gave you. It can be found right after a lock.

G4JFS, 2020-06-23:
Found it! Not particularly easy to spot as the console output window displays so much information. Is that information written out to a log file somewhere? I haven't found one yet. 

EId an important element for confirming the identification of a multiplex. I had assumed it would be displayed  in a more prominent place. Could it be?

andimik, 2020-06-25:
Although the Ensemble (long Label (in your case 'BBC National DAB') is a text field, it is a better indicator than the Ensemble ID.

Because in Europe different muxes are using same IDs.

I know it from Austria, within 1 hours drive I can receive Italian and Slovak ensemble, but both have the same ID causing mismatch in my car radio.

basicmaster, 2020-06-27:
One purpose of the EId is to be unique in a certain area (at least combined with the ECC), so there shouldn't be any ID used more than once. I can't find two similar EIds listed on the ukwtv.de....can you post an example?

andimik, 2020-06-27:
RAI and Slovak ensemble

basicmaster, 2020-06-27:
They have indeed the same EId. However, this is allowed - in combination with the ECC, the ensemble is unique, as Italy uses `0xE0` and Slovakia `0xE2` (as DAB defines). It is just a bit an unfortunate choice.

andimik, 2020-06-27:
The Czech CRo DAB use ID 0x2005 in the same country, but with different content.

And TELEKO has 0x2001 for both (different) multiplexes.

AlbrechtL, 2020-07-02:
Does it make sense to put the ensemble ID into the "Service Details" view?

G4JFS, 2020-07-02:
Would it not be better in Service Overview?

andimik, 2020-07-02:
As long it is only visible in the expert view.

G4JFS, 2020-07-05:
Expert view is certainly the place to put it.

AlbrechtL, 2020-07-11:
OK. This feature could be something for the version 2.3.
Pull request are welcome!

G4JFS, 2020-07-11:
Thanks. I look forward to it (in hex please!). I'm new here; where is the best place to put suggestions for the next version?

AlbrechtL, 2021-05-22:
Postponed to a later version. Pull request are still welcome.

AlbrechtL, 2024-10-09:
Added in https://github.com/AlbrechtL/welle.io/commit/6a1adf03c2acc66d7347ad18f43974b3bc075e70.

[image]

closed 2024-10-09 by AlbrechtL