AlbrechtL/welle.io #570: [Raspberry Pi] Mute button disappears when muting.

FartherOut, 2020-07-09:
[image]
[image]

As shown here, the speaker icon completely disappears when muting, making it difficult to unmute if one does not already know where the (invisible) button is.
I believe this is a simple icon issue, since the macOS version properly shows a red speaker icon when muted.

andimik, 2020-07-09:
Which version are you using?

FartherOut, 2020-07-09:
It's 2.2-dev I believe

AlbrechtL, 2020-07-11:
Funny I never experienced it on a PC or Android. Which display resolution do you have?

FartherOut, 2020-07-13:
It is the official Raspberry Pi 7" Touch Display
[https://www.raspberrypi.org/products/raspberry-pi-touch-display/](url)

Resolution is 800x480

Jasoroony, 2020-12-24:
I had to change the source code (in RadioView.qml) to remove the "Red" version of the "speakerIconMuted" item and switch everything over to just use the black one, to get this working on my Pi4 with QT5 -platform vnc

Interestingly though, when I used -platform linuxfb I didn't notice any issues.

I thought it was possibly just an issue in the QT vnc backend, but now I'm not so sure.  I'm using the latest snapshot of the code here, and QT 5.12.4

(With all the trouble I had getting QT to compile on my Pi4 I'm not inclined to rebuild to the latest version.)

AlbrechtL, 2021-03-27:
I tried the official Raspberry Pi 7" Touch Display and I observed similar issues with the raspberry OS stock Qt version. I have no idea where this is coming from. I also guess that this is more an Qt issue or graphic driver issue then a welle.io issue.

AlbrechtL, 2021-05-22:
I'm closing this issue for now because I don't see a solution. I anybody have an idea I will reopen the it.

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL