AlbrechtL/welle.io #572: [Raspberry Pi] Material theme menu is transparent.

FartherOut, 2020-07-09:
[image]
Material theme showing the GUI.

[image]
Material theme showing the settings menu.
Transparency issue makes it almost impossible to read the text and push the buttons.

Material theme works normally in the macOS version.

AlbrechtL, 2020-07-11:
Thanks for reporting!

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL