AlbrechtL/welle.io #573: 32kHz and 24kHz sampling rate not supported

andimik, 2020-07-12:
When an ensemble uses (both) 32kHz and 48kHz sampling rate (or 24kHz and 48kHz like in UK using MPEG 2.0 Layer II instead of MPEG 1.0 Layer II) then there will be no sound after you click on the first 32kHz (resp. 24kHz) service. It even will not play a sound after switching back to a 48kHz service.

DLS and SLS are working normally.

The console does not give a hint why.
[nosound_welle.txt](https://github.com/AlbrechtL/welle.io/files/4908261/nosound_welle.txt)
[nosound_welle_brisbane.txt](https://github.com/AlbrechtL/welle.io/files/4908265/nosound_welle_brisbane.txt)
[nosound_welle_uk.txt](https://github.com/AlbrechtL/welle.io/files/4908273/nosound_welle_uk.txt)
[nosound_welle_swr.txt](https://github.com/AlbrechtL/welle.io/files/4908282/nosound_welle_swr.txt)

I've checked with muxes from Slovenia, Australia, Germany and UK.

In Germany you cannot listen to SWR2 at all, for example.

See attached ETI, write a raw file with ODR-DabMod and play that raw
[swr_n.eti.zip](https://github.com/AlbrechtL/welle.io/files/4908284/swr_n.eti.zip)

All my other tools (Qt-DAB, dablin) play all stations with all 3 different sampling rates.

AlbrechtL, 2020-07-12:
Thanks for reporting.

AlbrechtL, 2020-07-12:
Please provide a RAW file for testing.

andimik, 2020-07-14:
It's very easy to create a raw file with ODR-DabMod, the command is only one line.

If you wish, I can upload such a file in the next days.

andimik, 2020-07-16:
Attached raw file was created with `$ odr-dabmod /tmp/swr_n.eti -f /tmp/swr_n.raw -F u8`
and truncated to 100 MB, then zipped

https://we.tl/t-RkgpLYv4BG

Tried under Ubuntu 20.04 and Ubuntu MATE 20.04 and `20200711_9f4342b_Linux_welle-io-x86_64.AppImage`

andimik, 2020-07-21:
@AlbrechtL

Please download the file till tomorrow as I won't upload it again.

If not, you still can create it with the simple line I provided.

orryverducci, 2020-09-12:
I'm seeing the same issue.

Here's a 60 second off air recording (u8 IQ format) of the Digital One National multiplex as broadcast on 11D here in the UK:
https://1drv.ms/u/s!AqWfaTxhIQ3zhY9HtVxLaGBKESLeFQ?e=otrcoF

It contains a mix of DAB services using 48kHz and DAB+ services using 32kHz.

If you select a 48kHz service first it will play audio along with other 48kHz services, but if you change to a 32kHz service the audio stops working for all services. The reverse is also true, so if you listen to a 32kHz service first it will play audio but when you change to a 48kHz service the audio stops working.

You can easily replicate the issue by listening Capital UK, which will play using a 32kHz sample rate, and then switching to KISS which is 48kHz. At the point of switching audio will stop playing for all stations, with no obvious errors.

AlbrechtL, 2020-09-19:
Thanks for the file! We will analyze it.

AlbrechtL, 2020-09-26:
I fixed something in https://github.com/AlbrechtL/welle.io/commit/c8515af840f85b0fd43cbe6e7fa40044c244a118. Please try again the nightly built 20200926_xxxx from https://bintray.com/albrechtl/welle.io/welle.io_nightly#files.

orryverducci, 2020-09-27:
I can confirm it's now working, thanks @AlbrechtL!

closed 2020-09-27 by AlbrechtL

andimik, 2020-10-01:
Confirmed, now it works.