AlbrechtL/welle.io #574: [macOS] Travis build failed for 9f4342b

AlbrechtL, 2020-07-12:
The macOS Travis CI build failed for commit 9f4342b.
See log: https://travis-ci.org/github/AlbrechtL/welle.io/jobs/707200144

@orryverducci Can you have a look, please?

orryverducci, 2020-07-13:
It seems in the last few days Homebrew has started installing Python 3.8 (it was previously 3.7), and this has broken the build and install for SoapySDR.

It's an upstream issue with the Pothos Homebrew formulas, I've opened an issue there (https://github.com/pothosware/homebrew-pothos/issues/41).

I'll see if this can be worked around, but I'm hoping the issue will be fixed.

AlbrechtL, 2020-07-13:
Thanks for checking!

orryverducci, 2020-07-27:
This has been fixed upstream. I've just run a build in Travis and it builds successfully.

AlbrechtL, 2020-07-27:
Great! A new binary is created https://bintray.com/albrechtl/welle.io/download_file?file_path=20200727_da5f3aa_MacOS_welle-io.dmg! Thanks!

closed 2020-07-27 by AlbrechtL