AlbrechtL/welle.io #577: Waterfall diagram for 16bit files broken

andimik, 2020-07-14:
Waterfall diagram for 16bit files seems not to work properly (= I just saw it once)

[image]
[image]

AlbrechtL, 2020-09-03:
Thanks for reporting.

AlbrechtL, 2020-09-30:
Can you provide a 16-bit IQ file please?

andimik, 2020-09-30:
Sure, please find it under

https://we.tl/t-5I2Ng26iks

It's Mux I Austria from Dobratsch on 6A.
Recorded with Qt-DAB, so it should be 16le

Download accessible within 7 days

AlbrechtL, 2020-10-01:
Thanks!
Fixed in https://github.com/AlbrechtL/welle.io/commit/6315838855c4f8abac4523737a79fa991561b829.

closed 2020-10-01 by AlbrechtL

andimik, 2020-10-01:
Confirmed, now it works