AlbrechtL/welle.io #581: SLS shown for previous service

andimik, 2020-07-14:
When I chose another service the slideshow still is shown for the previous service (I guess this only works correctly when the new service has SLS as well)

[image]

tenzap, 2020-07-16:
duplicate of #547 ?

andimik, 2020-07-16:
No, in my opinion not exactly the same, but similar.

I've to investigate more, but in my case the 2nd station has no SLS at all.

AlbrechtL, 2020-09-03:
Thanks for reporting. I keep both issues open until we analyzed the problem.

AlbrechtL, 2024-10-09:
I have issues to reproduce this issue. If a new service is selected, the MOT picture buffer will be cleared. Maybe it is fixed by accident in the last 4 years, or I didn't find it.

For now, I'm closing it.

closed 2024-10-09 by AlbrechtL