AlbrechtL/welle.io #585: 2.2-dev

Observerist, 2020-07-17:
Tried the **20200707_e6bd2f5_MacOS_welle-io** version, I don't usually go to seek nightlies, but there is one visible enhancement I like : in the Material/Dark style, station list is now white, way more readable.

Besides the white on white path of the SLS visible in expert mode, didn't discover anything else done to the GUI since v2.1. 

But just this white on grey lettering is already nice to have. ( still "praying" for a station list grouped by Muxes )

Thank you

[image]

AlbrechtL, 2020-09-03:
Thanks for reporting!

AlbrechtL, 2021-05-22:
The path was removed in https://github.com/AlbrechtL/welle.io/commit/6170d464c4affa2086bea41a4d6395651cff932e.

closed 2021-05-22 by AlbrechtL